emsec/ChameleonMini #123: Request for Cryptographic Function

shinestarqiao, 2017-07-03:
Hi, I have bought ChameleonMIni and want to do something about cryptographic.
Do you have some library for popular encryption methods, like AES and DES?
Thanks a lot.

geo-rg, 2017-07-06:
Have a look at the [fork of dev-zzo](https://github.com/dev-zzo/ChameleonMini/). For DESFire emulation, he has 3DES implemented.

I'm curious, what are you planning to do? :)

shinestarqiao, 2017-07-06:
Thanks for your help. Actually I want to implement some encryption codes into the tag to measure the delay time.

geo-rg, 2017-07-12:
So, how exactly can I help you? Or did my answer already answer your question fully?

geo-rg, 2018-09-12:
@shinestarqiao I suppose your question has been answered. If not so, please re-open this issue.

closed 2018-09-12 by geo-rg