emsec/ChameleonMini #124: XMODEM

metalparts, 2017-07-07:
I have problem when I am downloading the card to get the data .
Is NO data on the file always is only FF FF FF FF FF . 
I was using the command  DUMP_MFU and all the time i am getting message TIMEOUT .
please help.

Pepe

geo-rg, 2017-07-11:
Hi @metalparts 
`DUMP_MFU` can only dump MiFare Ultralight cards, I suppose that it does not work, because you try to dump anything other than a MiFare Ultralight card(?).

Then, if I understood correctly, you type `DOWNLOAD` and expect some card data to be downloaded. If `DUMP_MFU` was successful, it would write the contents of the MiFare Ultralight card to the terminal. If not, the timeout occurs. But in no case the card data is stored to the internal storage. Thus, you only receive dummy data.

metalparts, 2017-07-30:
Hi geo-rg,
Yes You r right about the Mifare Ultralight. I was trying to dump Mifare Classic 1K and Mifare Plus.

Then I was trying to download the content of the card via XMODEM and after to upload to some of the 8 ports in the Chamileon but I was not getting the right data to emulate the card .
Please let me know if I am mistaking something  .

Thank You

geo-rg, 2018-09-12:
@metalparts Even though I hope you already have found a solution: When running "download" you will NOT get the content of the card that is lying on the Chameleon, you will get the Chameleon's current emulation storage. With the Chameleon it is currently not possible to read Mifare Classic cards' content (hardware does support this, software not). But if you have the dump of a card, you can put it on the Chameleon via "upload" and set "config=MF_CLASSIC_1k" or whatever type of MFC card you want to emulate.

closed 2018-09-12 by geo-rg