emsec/ChameleonMini #127: USB stack problems

geo-rg, 2017-07-26:
tl;dr: Have a look at the end of this issue.

Currently we are using the LUFA CDC implementation for the USB interface which is for xmegas still experimental. We have experienced the following problems with that:
1. When using the Chameleon as reader via the command `send` with a PC-side running python script, communication between card and Chameleon sometimes breaks down.
2. When reading out the Chameleon as MF classic emulation, sometimes a sector cannot be read out.

We then have simply detached the Chameleon from the USB port and again tested both (for the first one, we tested a pre-programmed procedure) again and they worked nearly flawlessly. Also, we have seen on the scope, that when using the USB interface, sometimes a miller pulse sent from the Chameleon is a little bit too late. We conclude that the USB interface (and thus LUFA) sometimes breaks the timing while sending (or possibly also receiving) data.

We also tried to use the Atmel USB stack which made everything much worse. Also, on the ATxmega128A4U datasheet, there are no errata regarding the USB module.

So there are two messages for you to know:
1. **If you know anything we could do to solve this** or have any suggestions, it would be great if you could write an answer here.
2. If you want to emulate a card and eliminate the possibility of the USB interface breaking the timing, **please use the Chameleon in battery mode** without being connected to a PC.

Peterthegreat, 2018-04-05:
Any update on this issue? Using live mode gives bad nonces (compared to battery mode)

geo-rg, 2018-04-09:
We've had a big improvement with a recent update of the reader codec, which did not fix this problem completely, but has improved the reader ability (even with USB connection) a lot. Do you use the most recent version?

Peterthegreat, 2018-04-09:
Yes, I did check the latest version first. But I have to fix issue #131 first as I have a pretentious reader. I will get back to it in a few days. Thanks for the update