emsec/ChameleonMini #128: setting & Dump Mifare Classic 4K

milcaca, 2017-07-28:
I Have 2 issues the first one is that I can't set the slot via dialing through socat : 
```socat - /dev/ttyACM0,crlf
SETTING=2
200:UNKNOWN COMMAND
SETTING?
101:OK WITH TEXT
0
SETTING=?
202:INVALID PARAMETER
setting=2
root@kali:~/ChameleonMini/Software# 
```
Any reason why ? 

In addition to that I'd like to know if it's possible to dump a classic mifare 4K data from the reader ? 
 
thanks for any help

Peterthegreat, 2017-07-30:
Hi Mica,

Which command do you use for socat? Try "socat -,escape=0x0f /dev/chameleon,rawer,crnl" . 

[opens an interactive connection via the serial line, e.g. for talking with a modem. rawer sets the console's and ttyS0's terminal parameters to practicable values, crnl converts to correct newline characters. escape allows terminating the socat process with character control-O. Consider using READLINE instead of the first address.](http://www.dest-unreach.org/socat/doc/socat.html)

I don't know if you can read anything besides ultralight. Maybe you can get some data using the advance commands, but I did not tried (that hard) and never managed to obtain any data.

geo-rg, 2017-07-31:
Hi @milcaca,
the firmware does not yet support reading/dumping MF Classic cards, but it has all the functions you would need to build something that would dump a card (when the keys are given).

Did the hint from @Peterthegreat solve your other problem?

milcaca, 2017-08-19:
I managed to make it working, thank you for the help.
I have another question. I uploaded the provided MF classic 4K dump and tried to read it with a reader and that was not possible. Error say connection ended. Any idea of the reason ?

Peterthegreat, 2017-08-19:
Please provide the log from the emulator. (Using software/chamlog.py)

milcaca, 2017-08-21:
I'm trying to reupload the dump for verification but it always wails with this error 
```
[upload] 3584 Bytes successfully read from ../Dumps/MFC4k_randomContent_fixedKeys.mfd
Traceback (most recent call last):
  File "./chamtool.py", line 205, in <module> main()
  File "./chamtool.py", line 186, in main result = cmdFuncs[cmd](chameleon, arg)
  File "./chamtool.py", line 21, in cmdSetting result = chameleon.cmdSetting(arg)
  File "/root/ChameleonMini/Software/Chameleon/Device.py", line 209, in cmdSetting return self.getSetCmd(self.COMMAND_SETTING, newSetting)
  File "/root/ChameleonMini/Software/Chameleon/Device.py", line 165, in getSetCmd return self.writeCmd("{}{}{}".format(cmd, self.SET_CHAR, arg))
  File "/root/ChameleonMini/Software/Chameleon/Device.py", line 126, in writeCmd status = self.serial.readline().decode('ascii').rstrip()
UnicodeDecodeError: 'ascii' codec can't decode byte 0xd0 in position 2: ordinal not in range(128)
```

milcaca, 2017-08-21:
Ok I finally got the log (previous issue just stop at reboot ...)
I get this : 
```15031 ms <+15031 ms>:CODEC RX         (1   bytes) [26]
15031 ms <    +0 ms>:CODEC TX         (2   bytes) [4200]
15032 ms <    +1 ms>:CODEC RX         (2   bytes) [9320]
15032 ms <    +0 ms>:CODEC TX         (5   bytes) [7caa732287]
15034 ms <    +2 ms>:CODEC RX         (9   bytes) [93707caa732287444a]
15034 ms <    +0 ms>:CODEC TX         (3   bytes) [1837cd]
15058 ms <   +24 ms>:CODEC RX         (4   bytes) [500057cd]
15058 ms <    +0 ms>:APP HALT         (0   bytes) []
15061 ms <    +3 ms>:CODEC RX         (1   bytes) [52]
15061 ms <    +0 ms>:CODEC TX         (2   bytes) [4200]
15062 ms <    +1 ms>:CODEC RX         (9   bytes) [93707caa732287444a]
15062 ms <    +0 ms>:CODEC TX         (3   bytes) [1837cd]
15068 ms <    +6 ms>:CODEC RX         (4   bytes) [500057cd]
15068 ms <    +0 ms>:APP HALT         (0   bytes) []
15071 ms <    +3 ms>:CODEC RX         (1   bytes) [52]
15071 ms <    +0 ms>:CODEC TX         (2   bytes) [4200]
15072 ms <    +1 ms>:CODEC RX         (9   bytes) [93707caa732287444a]
15072 ms <    +0 ms>:CODEC TX         (3   bytes) [1837cd]
15077 ms <    +5 ms>:CODEC RX         (4   bytes) [500057cd]
15077 ms <    +0 ms>:APP HALT         (0   bytes) []
15081 ms <    +4 ms>:CODEC RX         (1   bytes) [52]
15081 ms <    +0 ms>:CODEC TX         (2   bytes) [4200]
15082 ms <    +1 ms>:CODEC RX         (9   bytes) [93707caa732287444a]
15082 ms <    +0 ms>:CODEC TX         (3   bytes) [1837cd]
15090 ms <    +8 ms>:CODEC RX         (4   bytes) [60017c6a]
15090 ms <    +0 ms>:APP AUTH         (2   bytes) [6001]
15091 ms <    +1 ms>:CODEC TX         (4   bytes) [a7294388]
15093 ms <    +2 ms>:CODEC RX         (8   bytes) [30cc7d895a5d3fc1]
15093 ms <    +0 ms>:APP AUTHING      (4   bytes) [43500ee2]
15093 ms <    +0 ms>:APP AUTH FAILED  (0   bytes) []
15098 ms <    +5 ms>:CODEC RX         (4   bytes) [500057cd]
15101 ms <    +3 ms>:CODEC RX         (1   bytes) [52]
15101 ms <    +0 ms>:CODEC TX         (2   bytes) [4200]
15103 ms <    +2 ms>:CODEC RX         (9   bytes) [93707caa732287444a]
15103 ms <    +0 ms>:CODEC TX         (3   bytes) [1837cd]
15108 ms <    +5 ms>:CODEC RX         (4   bytes) [500057cd]
15108 ms <    +0 ms>:APP HALT         (0   bytes) []
15111 ms <    +3 ms>:CODEC RX         (1   bytes) [52]
15111 ms <    +0 ms>:CODEC TX         (2   bytes) [4200]
15113 ms <    +2 ms>:CODEC RX         (9   bytes) [93707caa732287444a]
15113 ms <    +0 ms>:CODEC TX         (3   bytes) [1837cd]
15118 ms <    +5 ms>:CODEC RX         (4   bytes) [500057cd]
15118 ms <    +0 ms>:APP HALT         (0   bytes) []
15121 ms <    +3 ms>:CODEC RX         (1   bytes) [52]
15121 ms <    +0 ms>:CODEC TX         (2   bytes) [4200]
15123 ms <    +2 ms>:CODEC RX         (9   bytes) [93707caa732287444a]
15123 ms <    +0 ms>:CODEC TX         (3   bytes) [1837cd]
15128 ms <    +5 ms>:CODEC RX         (4   bytes) [500057cd]
15128 ms <    +0 ms>:APP HALT         (0   bytes) []
15131 ms <    +3 ms>:CODEC RX         (1   bytes) [52]
15131 ms <    +0 ms>:CODEC TX         (2   bytes) [4200]
15133 ms <    +2 ms>:CODEC RX         (9   bytes) [93707caa732287444a]
15133 ms <    +0 ms>:CODEC TX         (3   bytes) [1837cd]
15138 ms <    +5 ms>:CODEC RX         (4   bytes) [500057cd]
15138 ms <    +0 ms>:APP HALT         (0   bytes) []
15141 ms <    +3 ms>:CODEC RX         (1   bytes) [52]
15141 ms <    +0 ms>:CODEC TX         (2   bytes) [4200]
15143 ms <    +2 ms>:CODEC RX         (9   bytes) [93707caa732287444a]
15143 ms <    +0 ms>:CODEC TX         (3   bytes) [1837cd]
15148 ms <    +5 ms>:CODEC RX         (4   bytes) [500057cd]
15148 ms <    +0 ms>:APP HALT         (0   bytes) []
15151 ms <    +3 ms>:CODEC RX         (1   bytes) [52]
15151 ms <    +0 ms>:CODEC TX         (2   bytes) [4200]
15153 ms <    +2 ms>:CODEC RX         (9   bytes) [93707caa732287444a]
15153 ms <    +0 ms>:CODEC TX         (3   bytes) [1837cd]
15158 ms <    +5 ms>:CODEC RX         (4   bytes) [500057cd]
15158 ms <    +0 ms>:APP HALT         (0   bytes) []
15161 ms <    +3 ms>:CODEC RX         (1   bytes) [52]
[... 194 lines cut ...]
19091 ms <    +1 ms>:CODEC RX         (9   bytes) [93707caa732287444a]
19091 ms <    +0 ms>:CODEC TX         (3   bytes) [1837cd]
19222 ms <  +131 ms>:CODEC RX         (4   bytes) [500057cd]
19222 ms <    +0 ms>:APP HALT         (0   bytes) []
19225 ms <    +3 ms>:CODEC RX         (1   bytes) [52]
19225 ms <    +0 ms>:CODEC TX         (2   bytes) [4200]
19227 ms <    +2 ms>:CODEC RX         (9   bytes) [93707caa732287444a]
19227 ms <    +0 ms>:CODEC TX         (3   bytes) [1837cd]
19359 ms <  +132 ms>:CODEC RX         (4   bytes) [500057cd]
19359 ms <    +0 ms>:APP HALT         (0   bytes) []
19362 ms <    +3 ms>:CODEC RX         (1   bytes) [52]
19362 ms <    +0 ms>:CODEC TX         (2   bytes) [4200]
19363 ms <    +1 ms>:CODEC RX         (9   bytes) [93707caa732287444a]
19364 ms <    +1 ms>:CODEC TX         (3   bytes) [1837cd]
19494 ms <  +130 ms>:CODEC RX         (4   bytes) [500057cd]
19494 ms <    +0 ms>:APP HALT         (0   bytes) []
19497 ms <    +3 ms>:CODEC RX         (1   bytes) [52]
19497 ms <    +0 ms>:CODEC TX         (2   bytes) [4200]
19498 ms <    +1 ms>:CODEC RX         (5   bytes) [93707caa01]
19499 ms <    +1 ms>:CODEC RX         (2   bytes) [91c3]
20176 ms <  +677 ms>:CODEC RX         (1   bytes) [26]
20176 ms <    +0 ms>:CODEC TX         (2   bytes) [4200]
20177 ms <    +1 ms>:CODEC RX         (2   bytes) [9320]

```

Peterthegreat, 2017-08-21:
Looks like the emulation error that is present in the latest firmware. I notice the bug but I am currently out of town and cannot identify the exact commit where the issue appeared.

You can try tree [05fa61c](https://github.com/emsec/ChameleonMini/tree/05fa61c7db0ad4cb32dc52edaba9737f1d41983a)

But you must first compile the source. The .hex and .eep files will appear in ChameleonMini/Firmware/Chameleon-Mini/

milcaca, 2017-08-21:
I'll try that, thanks for the help