emsec/ChameleonMini #130: AUTOCALIBRATE

metalparts, 2017-07-30:
Hello,
there is a problem in a Reader Mod  when I execute the command  AUTOCALIBRATE:
The answer is all ways UNKNOWN COMMAND.
Please let me know how to fix this.

Thank You

Peterthegreat, 2017-07-30:
Hi,
First, you must set the Chameleon to reader mode using the config command. Then, simply run AUTOCALIBRATE

geo-rg, 2017-07-31:
Hi @metalparts 
if the answer is `UNKNOWN COMMAND`, you may have an older version of the firmware, which does not yet know this function. If the answer is `INVALID COMMAND USAGE`, you would have to set the configuration to `ISO14443A_READER`.

metalparts, 2017-07-31:
Hi Peterthegreat
The Chameleon is on reader mod and I am sending the others readers command and they work fine  but when i am trying AUTOCALIBRATE  the answer is UNKNOWN COOMAND 
Thank You

metalparts, 2017-07-31:
Hi geo-rd,
My version is ChameleonMini RevG 170614 using LUFA 151115 
compiled with AVR-GCC 4.9.2.
commit 2a2e074
The Chameleon is in reader mode when I send command THRESHOLD? I receive Answer  OK 400.
When I send AUTOCALIBRATE I am receiving answer 200 - UNKNOWN COMMAND.
Please let me know How to fix this.
Thank You

geo-rg, 2017-07-31:
You need to upgrade your firmware to the latest firmware.

Peterthegreat, 2017-07-31:
type help and look for autocalibrate command

metalparts, 2017-07-31:
Hi Peterthegreat,
thank you for your answer I type HELP and I don't have the AUTOCALIBRATE with the others commands.
 I will upgrade to the latest version.

metalparts, 2017-07-31:
HI geo-rd,
thank you I will do this.

metalparts, 2017-07-31:
Hi geo-rg,
thank you I upgrade the version and now I have Autocalibrate.
Regards

closed 2017-07-31 by metalparts