emsec/ChameleonMini #133: Logging KeyA and KeyB requests

yaba, 2017-08-16:
I'm studying a system that generates KeyA and KeyB based on UID.
Is Chameleon Mini capable of generating a couple of sequential UIDs and log the keys used to read the sectors (except KeyA from sector0 since this is a fixed one)?
For example, let's say I know that for UID 00112233 machine will use KeyA 010203040506.
I need to understand what key the machine will use if I use pre-defined UID like 11223344 and log that that request.

emsec, 2017-08-16:
ChameleonMini can generate different UIDs and log the communication, you can a) control it from a PC e.g. with a Python script (see python folder in the chameleon github project for an example script)  or b) change the firmware accordingly but I think a) is the better option for you. 
 
configure as CONFIG=MF_CLASSIC_1k and then use the UID command to set the UID, e.g., UID=11223344
 
Next, use the chamlog tool in the github project to get the log in human-readable form.

However, note that cryptographic keys are NEVER sent during an authentication (that's the whole idea of a Challenge-Response protocol as used in the Mifare Classic cards...) so you cannot log the keys directly, but for each new UID you would have to crack the keys, somehow....  (It is technically feasible to find the keys from logging the communicaion but this topic is too big to be discussed here...)

Peterthegreat, 2018-04-05:
@yaba I was working on this a while back. PM me as I don't want to publish the code just yet.

yaba, 2018-04-05:
@Peterthegreat github doesn't allow PM anymore, can I use the emsec e-mail ?

Peterthegreat, 2018-04-05:
@yaba Guess so. I have no idea how that works.

leperjulien, 2018-04-06:
@Peterthegreat 
Actually , i might also be interested if you don't mind !
That would be a great opportunity to take my ChameleonMini out of the drawer since it has been laying there for quite some time now.

geo-rg, 2018-09-12:
@yaba I suppose everything is clear now, if not so, please re-open this issue.

closed 2018-09-12 by geo-rg