emsec/ChameleonMini #134: External antenna for emulation little tag

etmatrix, 2017-08-28:
Can I use an external antenna and replace PCB? I need emulate a mifare about 2cm (0.79 inches), ChameleonMini is a good choice for that scope?

geo-rg, 2017-08-29:
Hi @etmatrix,
adjusting the antenna is very complex and we can give no guarantees that emulation or any other functionality will work after your changes. However, it is possible that it could work.

kaliginium, 2018-01-24:
Hi,

 do you know what type is exactly connector on chameleon board?

Endres, 2018-01-24:
It is a standard U.FL plug, so you will need a U.FL jack (usually mounted on a pigtail-style-cable) for connecting to the antenna signal. It is the same type of plug used in modern laptop computers for the Wi-Fi antenna.

kaliginium, 2018-01-31:
Thank you,

 I order connector, make antenna a try if antenna is working on proxmark. Then I connect antenna to chameleon and it did not working. I there needed to somehow switch which antenna either internal oe external should be used?

geo-rg, 2018-02-02:
Hi @kaliginium 
it is not trivial / highliy difficult to use another antenna. Note that the whole circuitry is adjusted to the standard antenna. Any small change on the antenna may change the signal arbitrarily and thus may break functionality.

kaliginium, 2018-02-09:
I note that. I hope I can calculate that, but there is the trouble, that I do not know the circuit. I have trouble, that my application is able to comunicate with chameleon only in distance cca 9-13 cm from reader. I rewrite FW to log me, what is happen and chameleon response in standard way. But on standard card is reader responsible only in distance 2-3 cm. So maybe I am not able to emulate a card. Thus I am about to create external antenna to maybe solve this problem.

geo-rg, 2018-03-22:
OK, @kaliginium, let us know if you have success, I hope you can solve your problem. However, do you have any remaining (and related) questions? If so, please re-open this issue. Same for @etmatrix.

closed 2018-03-22 by geo-rg