emsec/ChameleonMini #135: Cloning classic 4k

milcaca, 2017-08-29:
Before trying to get my hands dirty and try to do something would it be possible to use the clone comand to dump a MF4K/1K if the keys are known ? 
If so any clue of where I should look into the code to get some point to start ? 
regards #

geo-rg, 2017-08-29:
Currently this is not implemented. The necessary Crypto1 functions are implemented, but the application layer for reading MF Classic is not. Just have a look at [these lines](https://github.com/emsec/ChameleonMini/blob/master/Firmware/Chameleon-Mini/Application/Reader14443A.c#L557-L624) for a first start.

milcaca, 2017-08-29:
thanks for the info, I think I'll have a lot of work to do :/
BTW, I have a ASCII style dump of NFC MF4 card, I'd like to convert or create a file to have a .mfd file in order to upload to Chameleon. Any idea ?

geo-rg, 2017-08-29:
I think xxd is useful for this.