emsec/ChameleonMini #137: Sniffing - Logging question

tzanou, 2017-09-02:
Can chameleonMini sniff - log communication between a reader and a Mifare Ultralight Ev1 ?

geo-rg, 2017-09-18:
Yes, but only the reader -> card direction, not the other way round.

lviarani, 2017-11-13:
Hello to everyone, I'm brand new to github. I've learnt today of this issue, is there any chance of having the two-way sniff (PICC <> PCD)?
I mostly use the Chameleon rev. G as a sniffer to debug NFC, I'm in trouble without messages "PCD > PICC" (i.e. from the reader to the card)...

One further question: please take a look at the attached files, if instead of chamlog I just use minicom it seems some more bytes are regularly caught... is that noise or some very-low-level RF info or could I expect that if I use minicom I can log also messages "PCD > PICC"?

My setup is made of a NFC reader sending the REQA and a Mifare classic card.
I'm using ubuntu, minicom and chamlog.py

Thank you very much.

[reader with no card-chamlog.txt](https://github.com/emsec/ChameleonMini/files/1467691/reader.with.no.card-chamlog.txt)
[reader with no card-minicom.txt](https://github.com/emsec/ChameleonMini/files/1467692/reader.with.no.card-minicom.txt)

hydra228, 2018-01-22:
why does not full duplex work? how to solve the possibility of transmitting PICC <-> PCD is a demo limitation or what?

geo-rg, 2018-01-23:
Reader->Card is basically the same codec like emulation, since emulation also needs to decode the reader's request when the Chameleon itself does not generate a field. However, the DEMOD signal for the other direction (card->reader) is different from the signal in the reader codec, since in the reader codec additionally the Chameleon generates a field. As a consequence, the sniffing codec currently is not able to decode the card's answers.

hydra228, 2018-01-24:
Did I understand correctly - chamelion supports card <-> terminal hardware, but does not support software? Will you try to fix it later?

geo-rg, 2018-01-24:
The Chameleon **hardware** _should_ support sniffing also in the Card->Reader direction, as I have seen some promising pictures from signals that could be used for this. However, I cannot give any guarantee whether or when it will be implemented. It's not about fixing something, it's about implementing this feature for the first time.

gypsophlia, 2018-08-15:
Hi, I have written some code for the chameleon to sniffing in both ways. Please feel free to have a try: #180 #181