emsec/ChameleonMini #138: Writing to Chameleon with another reader

tempuser1909, 2017-09-03:
Hi, I was wondering if Chameleon supports emulation that can handle read/write from another reader rather from the direct serial/usb connection from a PC.

I can't seem to find any information regarding this. Hopefully someone can enlighten me.

Thank you.

geo-rg, 2017-09-18:
What exactly do you mean? Currently, you can only upload or download a card dump to/from the Chameleon.

maxieds, 2018-01-04:
_Another shameless plug:_ There is also now an open-source Android app available on the [Play Store](https://play.google.com/store/apps/details?id=com.maxieds.chameleonminilivedebugger) that can read live logging data, execute shell commands, and read from the Chameleon Mini device. Requires a usb adapter to get it hooked up with the standard cable.

jtl21, 2018-01-05:
Available in Canada?

doegox, 2018-01-05:
Hi @maxieds what are the device/Android/country requirements to be able to get it from the Play Store?
It's told to be incompatible with my various devices, while I guess all I need is USB OTG.

maxieds, 2018-01-08:
Should now be available in Canada. There problem with rolling out releases per country is that you have to enable them one-by-one. The release works well on Android 7.0.1 (Lollipop). Is it possible you are using an older phone?

jtl21, 2018-01-11:
I'm on a s6 but i just rechecked and it's available for download now.
Thanks!!