emsec/ChameleonMini #140: Problems reading some smaller Mifare 1K RFID tags

ahzf, 2017-09-19:
First of all, thanks for this great project and product :)!
It works very well reading e.g. all those e-mobility "Ladekarten" used all over Europe https://twitter.com/OCCloud/status/906914234264702976 I hope those companies will get smarter soon.

But with reading the UID of a couple of smaller RFID tags like http://www.watterott.com/de/RFID-Tag-Transparent-MIFARE-1K-1356-MHz or http://www.watterott.com/de/Mifare-One-RFID-Tag-1356MHz fails. Any idea why this might happen? At least with an mfrc522 those tags can be read and a scan does not find  anything.

geo-rg, 2017-09-20:
Hi @ahzf 
first of all, thanks for your compliments!

The Chameleon indeed can have problems with reading non-credit-card-shaped tags. You could first try the new `autocalibrate` command, which tries to select a card with a big range of reader thresholds and gives you feedback (+ or -) for every threshold whether it worked or not. I would suggest you try this with your tags in different positions. E.g. right in the middle of the antenna should not work as good as when the tags antenna lies directly over the Chameleons antenna.
You may also try it with credit-card-shaped cards to get a feeling how much worse communication to a small tag is.

ctshh, 2018-01-17:
_I second this issue - very strongly so._
So far I am having mixed feelings about my ~150€ purchase:
- DESfire - advertised, yet not supported (yet?).
- **Small tags (Stadtrad, "wrist-tags") - no readout, `autocalibrate` gives only `-`'s.** This _needs_ to be improved, else you will leave very many customers quite unhappy. It's just too expensive to not read these tags. And the pictures you, the makers, show on your webpage do depict small wrist-type tags so every readers assumption will be that it can read them.
- Vatt****** Ladekarte - `Unknown card type.`, cannot be operated on.
Correct me if I'm wrong, but this device does not yet do it's job and surely needs a bit of polishing.
Please, by all means, do polish this device.

I know this sounds harsh. It's just text, emotions are interpreted by the reader. It's not meant that way. It's a promising device, but please - don't sell Bananenhardware.

geo-rg, 2018-01-18:
Hi @ctshh 
first of all, thank you for bringing your criticism politely. Another thing I have to mention before going into details is that an open source project always lives from those who actively contribute to the project.

Regarding DESfire: What do you want to do? There is a fork in which emulation is under development and many features have been finished by now. Reading DESfire, of course, needs software adjustments which on the other side are not that complex since they would be only on application layer. However, we only advertised the _hardware_ to support DESfire _in principle_ which means that there is still software work to do. That brings me to my introductory words that an open source project always lives from the people that contribute to it!

As for the small tags and the Ladekarte you wrote about, please be aware that the reader config only supports ISO 14443 type A. Many of the wrist tags are 125 kHz or ISO 15693 or others that are not yet supported. However, I will do my best to help you dealing with these cards/tags if you are sure that they are ISO 14443 type A.

I also would like to point out that the 150 € you have spent may seem much from your side, but from our side - as we are a small company - it covers production cost and initial development expenses. In the end, it is **non-profit** for us. We really do not sit here drinking champagne and laughing about people who are unsatisfied with the project. We have developed it with devotion and are happy for everyone who could realize projects with the Chameleon, learn something about NFC or about embedded programming.

ctshh, 2018-01-19:
Thank you, @geo-rg, for taking time to answer thorough and kind.

I now understand that I (as a not very gifted programmer and really not much of an developer) will have to wait for DESfire functionality.
I also understand the pricing better now.

As for the Stadtrad tag I can confirm it to be a Mifare classic 1k, as read by my trusted Arduino RFID reader. It is depicted here: https://stadtrad.hamburg.de/kundenbuchung/templates/510/layout/bilder/schluesselanhaenger.jpg

The Vattenfall Ladekarte is a Mifare classic with writable/unlocked data fields (I've successfully changed them a few years ago when my Ladekarte was new, so yes, the system only checks the UID - big news).
Is there a non-public way I can send you further Infos on the card?

doegox, 2018-01-19:
For very small tags, a large antenna is not adapted but sometimes you may have luck by building a very simple "relay attack": take a wire, about 3 or 4 large loops connected to 3 or 4 small loops.
This allows me e.g. to read these 4mm large Mifare tags (the pic was with another reader):

[image]

ctshh, 2018-01-20:
After having tried different gauges of wire and different numbers of windings I'm afraid I'm not having any success. Both iPhone 8 and a Samsung Galaxy A3 identify and read (in the case of the A3, also write) to the tag(s) I'm using here successfully. If you'd like one I can send you a tag via postal mail for your own tinkering.
Now, I was wondering if the little U.FL connector on the board might help? Can I connect some external antenna to it? If so, would you have a recommendation for an antenna?

maxieds, 2018-01-20:
I'm also interested in trying to write a Desfire card implementation for the Chameleon. Due to school and other commitments, however, it will probably be a good long *while* before I have a chance to look into this. The [android app](https://github.com/maxieds/ChameleonMiniLiveDebugger) I wrote has recognition for Desfire commands and status codes in the LIVE logging interface, mostly pulled from the nice Desfire HCE repo by @jekkos and [pp. 17-21 of this document](https://web.archive.org/web/20170201031920/http://neteril.org/files/M075031_desfire.pdf) ([see also this issue](https://github.com/jekkos/android-hce-desfire/issues/1)). I don't think that the linked document is easy to find, so anyone trying to actively create a Desfire card configuration / profile for the Chameleon might want to check it out.

geo-rg, 2018-01-22:
@maxieds Check out https://github.com/dev-zzo/ChameleonMini/tree/desfire which is a really good start for DESfire emulation.

@ctshh Have you tried positioning the tags directly on the antenna of the Chameleon and not only in the middle of it? However, (most likely) today I will push an upgrade to the ISO 14443A reader codec that fixes some problems and possibly could also help communicating with these smaller tags. What you could do anyways is to execute an autocalibrate while having `logmode=MEMORY` and then append the log here so I can see, what happens there.

maxieds, 2018-01-22:
> @maxieds <https://github.com/maxieds> Check out https://github.com/dev-zzo/
> ChameleonMini/tree/desfire which is a really good start for DESfire

Is there a good way to mock up a Chameleon in software to build directories
and files on a sample card. I'd like to generate a slightly more extensive
set of card dumps that can then be reloaded on the Mini by an upload via
XModem.

ctshh, 2018-01-31:
@maxieds 
> (...) However, (most likely) today I will push an upgrade to the ISO 14443A
reader codec that fixes some problems and possibly could also help
communicating with these smaller tags. (...) 

I hate "being that guy", but... any news on this? ;)

maxieds, 2018-01-31:
Can you be more specific? I wasn't following the thread that closely. I
know @gtpy has some desfire addons for my app at some point.

ctshh, 2018-01-31:
@maxieds @gtpy: I know about the DESfire situation, I was just inquiring about the reader update. I'll try this out as soon as I'm back home. Thank you!

maxieds, 2018-01-31:
The reader issues in any previous versions of my app should have been
fixed. (There were a few bugs in previous releases) Not sure about the
updates @gtpy is talking about.

gtpy, 2018-01-31:
That was update from firmware not at your app :-)

-------- Original-Nachricht --------

geo-rg, 2018-01-31:
Hi @ctshh 
I pushed the update the same day (or one day later). There have been changes on the ISO14443A_READER codec, which should improve the reading capability. If there are any problems with it, please report them ASAP. :)

ctshh, 2018-02-05:
@geo-rg I have installed the latest firmware, now I'm getting a rather strange behavior: `autocalibrate` and `identify` don't seem to work anymore, at least not as before, and they don't seem to show up in my `help`-list:

```
version?
101:OK WITH TEXT
ChameleonMini RevG 180125 using LUFA 151115 compiled with AVR-GCC 5.4.0. Based on the open-source NFC tool ChameleonMini. https://github.com/emsec/ChameleonMini commit c6d2968
help
101:OK WITH TEXT
VERSION,CONFIG,UID,READONLY,UPLOAD,DOWNLOAD,RESET,UPGRADE,MEMSIZE,UIDSIZE,RBUTTON,RBUTTON_LONG,LBUTTON,LBUTTON_LONG,LEDGREEN,LEDRED,LOGMODE,LOGMEM,LOGDOWNLOAD,LOGSTORE,LOGCLEAR,SETTING,CLEAR,STORE,RECALL,CHAE
identify
201:INVALID COMMAND USAGE
identify?
201:INVALID COMMAND USAGE
autocalibrate
201:INVALID COMMAND USAGE
autocalibrate?
201:INVALID COMMAND USAGE
```

Any ideas what I'm doing wrong? I'm still on my desperate path to read small tags ;)

I've attached a picture of the tag I'm trying to read, my Android phone (easily and repeatedly with a 0% error rate) identifies the chip as `NXP MIFARE Ultralight (Ultralight)`, `Tyg Typ: ISO 14443-3A` with Serial Number, ATQA and SAK visible. The Data fields are programmable, too.

[image]

gtpy, 2018-02-05:
@ctshh is your slot configured as ISO1443A_READER?

gtpy, 2018-02-05:
should look like this:
**version?**
101:OK WITH TEXT
ChameleonMini RevG 180125 using LUFA 151115 compiled with AVR-GCC 5.4.0. Based on the open-source NFC tool ChameleonMini. https://github.com/emsec/ChameleonMini commit c6d2968
**help**
101:OK WITH TEXT
VERSION,CONFIG,UID,READONLY,UPLOAD,DOWNLOAD,RESET,UPGRADE,MEMSIZE,UIDSIZE,RBUTTON,RBUTTON_LONG,LBUTTON,LBUTTON_LONG,LEDGREEN,LEDRED,LOGMODE,LOGMEM,LOGDOWNLOAD,LOGSTORE,LOGCLEAR,SETTING,CLEAR,STORE,RECALL,CHARGING,HELP,RSSI,SYSTICK,SEND_RAW,SEND,GETUID,DUMP_MFU,IDENTIFY,TIMEOUT,THRESHOLD,AUTOCALIBRATE,FIELD,CLONE
**config?**
101:OK WITH TEXT
NONE
**config=iso14443a_reader**
100:OK
**config?**
101:OK WITH TEXT
ISO14443A_READER

ctshh, 2018-02-05:
Ha! Stupid me!

```
101:OK WITH TEXT
MIFARE Ultralight
ATQA:   4400
UID:    0431xx52xx4Cxx
SAK:    00
```

I can hereby say that the current version _does_ read better. **Thank you!**

geo-rg, 2018-03-22:
So, @ahzf , are there any open questions? If so, please re-open this issue.

closed 2018-03-22 by geo-rg