emsec/ChameleonMini #142: Stuck with Firmware upgrade

learner4life, 2017-09-26:
I am using Windows 7 pro 64 Bit. Recently got Chameleon mini from 

https://www.aliexpress.com/item/Newest-13-56MHZ-ISO14443A-Chameleon-Mini-RDV2-0-Kits-RFID-Copier-Duplicator-UID13-56MHZ-NFC-Card/32820382051.html?spm=a2g0s.9042311.0.0.AKT7xs

I was able to install driver and now device is getting reflected on Com port as chameleon.

However, as i want to upgrade firmware , i get following error with Tera Term

200:UNKNOWN COMMAND

When i try to use firmware upgrade.bat, i am getting error 

There was an error with executing this command. Maybe your ChameleonMini is not in bootloader mode?

Please note that all required files are in same folder.

How do i resolve this?

Also, i have noticed chipset on device is ATXMEGA 32A4U AU 1449 and not atxmega128a4u

geo-rg, 2017-09-26:
I assume you have only noticed that your previous issue has been closed, but I also responded to it. Here again the answer:
Since the hardware seems to be completely different from our original hardware, we can give no support in this case. Please contact the manufacturer of the product you have bought.

learner4life, 2017-09-26:
My apologies. i didnt notice earlier closed thread.

Appreciate your revert. I bought this hardware assuming it is commercial version sold by you.

Where can i buy your genuine product?

Thanks

geo-rg, 2017-09-26:
(https://shop.kasper.it/chameleonmini)

closed 2017-09-27 by geo-rg