emsec/ChameleonMini #143: NTAG support

rctgamer3, 2017-09-27:
NTAG doesn't seem to be supported - filing a meta bug for NTAG support. Really hope NTAG215 emulation will be possible someday.

cyber-vi-king, 2022-09-19:
Hi,
this Issue seems to be the best place to ask this.

In #253 the functionality for NTAG215 seems to has been added.
I can indeed configure a slot with `config=NTAG215` and when using some NFC app on my iPhone, the Chameleon shows up as NTAG215.

My problem however - how do I get data onto this NTAG215?
With NFC Tools on iOS I cannot write directly to the Chameleon
And when I dump an NTAG215 (that I wrote with my iPhone) with the Proxmark3 RDV4 and upload said dump to the chameleon it won't work either -> It shows just garbage - and dumping it again with the Proxmark3 shows that the two dumps are completely different.

fptrs, 2022-09-21:
Hi @cyber-vi-king,
can you provide the dump and the log from the chameleon?

cyber-vi-king, 2022-09-21:
Sure thing, I'll dump it again tonight when I'm back at my computer. 

How do you want the data? Binary upload somewhere? Or the hex here as code snippet formatted?

fptrs, 2022-09-21:
I'd prefer a binary upload for the dump and yes that is the correct way regarding the log.

cyber-vi-king, 2022-09-21:
Bad News. I'm in a Hotel at the moment, so my main bag of fun stuff stayed at home.
I do have my Proxmark3 and Chameleon Tiny with me, but the only NTAG215 card I'm
carrying won't be read by the Proxmark3 - it does fine however with my iPhone and Android 
phone. Now I'll have find an android app that allows me to dump the NTAG215 as the NFC Tools app on iOS just shows me the data but does not provide a binary dump that I can upload to the chameleon - which due to reflashing firmware no longer stores the dump I initially tried.

I'll try to get a dump and link a gist here, with both the log and the dump as soon as I have it.

fptrs, 2022-09-22:
Hi @cyber-vi-king,
I'm sorry to inform you that we only offer support for the original Chameleon. I suggest you open an issue in the manufacturers repo.

closed 2022-09-22 by fptrs