emsec/ChameleonMini #144: Problem with ISO14443A-SNIFF mode logs, weird symbols

pexes, 2017-10-04:
Hi. I'm trying to get logs from my Chameleon-mini, but i'm getting set of unclear symbols. I have tried to update firmware, but it not helps. Help me please, i tried everything and you are my last chance. 
Here is my log
[log (2).txt](https://github.com/emsec/ChameleonMini/files/1357197/log.2.txt)

geo-rg, 2017-10-05:
I was just able to decode your log. You did everything correct, but skipped the last step.

You have downloaded the log correctly, but the Chameleon does not give it to you human-readable, but in a special format. You can convert this by using python. You can find `chamlog.py` in the ChameleonMini/Software folder. Usage at command line is: `python3 chamlog.py -f path/to/your/downloaded/logfile`. Depending on the system you use, the `python3` command may be not necessary or may be just `python`.

geo-rg, 2017-10-06:
@pexes Please close this issue if your problem is solved now.

closed 2017-10-10 by pexes