emsec/ChameleonMini #145: Any update regarding mifare desfireEv1 support?

vagha, 2017-10-05:


geo-rg, 2017-10-05:
There is the fork <https://github.com/dev-zzo/ChameleonMini/tree/desfire>, which covers a part of DESFire EV1, maybe the functionality you need is already there?

dev-zzo, 2017-11-17:
@vagha Please check the fork and open issues there; any help with those marked with "help needed" will be greatly appreciated.

closed 2018-01-08 by geo-rg

hamsik7, 2018-01-09:
@geo-rg  I have an issue, how to clone/dump Mifare DesFire ev1 any hints will help me alot
[image]

dev-zzo, 2018-01-09:
@hamsik7 Please use e.g. proxmark3 for that. You will need the card keys anyway, there is no way to dump anything without those. In any case, it is irrelevant to Chameleon.

hamsik7, 2018-01-09:
@dev-zzo is there any alternative to proxmark3 ? maybe a tool ? i found Mfcuk but i dont think its relevant !