emsec/ChameleonMini #146: compilation issue

kaliginium, 2017-10-12:
Hi,

can you help me with compilation? I still get error:
error: cannot access device-specs for 'atxmega128a4u' expected at 'device-specs/specs-atxmega128a4u'

thank you

geo-rg, 2017-10-16:
Hi @kaliginium 
Which OS and compiler are you using? It seems like your toolchain is configured wrong.

kaliginium, 2017-10-16:
I am using Windows 10 OS. Maybe toolchain is configured wrong as you say. I am trying to compile code in cygwin environment,

kaliginium, 2017-10-16:
Latest I get error:

as: unrecognized option `-mmcu=avrxmega7'
In file included from Chameleon-Mini.h:7:0,
                 from Chameleon-Mini.c:1:
./avr/power.h:1149:1: warning: function declaration isn't a prototype [-Wstrict-prototypes]
 __power_all_enable()
 ^
./avr/power.h:1198:1: warning: function declaration isn't a prototype [-Wstrict-prototypes]
 __power_all_disable()
 ^
Chameleon-Mini.c:37:1: fatal error: error writing to -: Invalid argument
 }
 ^
compilation terminated.
make: *** [../LUFA/Build/lufa_build.mk:295: Bin/Chameleon-Mini.o] Chyba 1

kaliginium, 2017-10-18:
Solved. Just toolchain configuration. Sorry for abuse.

geo-rg, 2017-10-18:
Glad you could solve it yourself, no need for excuses :)

closed 2017-10-18 by geo-rg