emsec/ChameleonMini #147: Is the SAK value changeable?

fener06, 2017-10-26:
I'm looking for a possibility to change the SAK value. In the original mifare 1k and the dump file of this card has a SAK value with 88. But after uploading the dumpfile, i'm getting the SAK value by reading 08.

Google says: 88 are Infenion Cards and 08 are NPX cards.

Is there a Option to change by chameleonmini the card manifacture?

THANKS

doegox, 2017-10-26:
There is no option at the moment but you can simply change it here and recompile the firmware:

https://github.com/emsec/ChameleonMini/blob/master/Firmware/Chameleon-Mini/Application/MifareClassic.c#L21

fener06, 2017-10-27:
Thanks! It has worked :) I think it would be usefull to implement the Option to change the SAK value and the ATQA value 👍

closed 2017-10-27 by fener06