emsec/ChameleonMini #148: flash worn out?

tned73, 2017-11-02:
I used the chameleonMini for regression testing, which means pushing a lot of MFClassic images to the board, well over 500.000 times and logging the trace and the changed image.
Both boards stopped functioning and re-flashing the firmware failed at validation. 

Reading | ################################################## | 100% 0.02s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0001
         0x02 != 0x00
avrdude: verification error; content mismatch

avrdude done.  Thank you.

Is my presumption right that it killed the flash memory due to all the images pushed towards the board? 
If so are there suggestions how to change the code to keep the image and logging in RAM and keep the flash untouched during runtime?

skuep, 2017-11-02:
Most probably not. What hardware did you use for flashing? Do you use ISP? What clock speed are you using?

tned73, 2017-11-02:
Hi Skuep,

Up till now I haven been able to flash through the USB interface from a linux system, I have not used a ISP.

tned73, 2017-11-02:
Complete flashing log:
>sudo avrdude -c flip2 -p ATXMega128A4U -B 60 -P usb -U application:w:Chameleon-Mini.hex:i -U eeprom:w:Chameleon-Mini.eep:i

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9746 (probably x128a4u)
avrdude: NOTE: "application" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "Chameleon-Mini.hex"
avrdude: writing application (131072 bytes):

Writing | ################################################## | 100% 1.61s

avrdude: 131072 bytes of application written
avrdude: verifying application memory against Chameleon-Mini.hex:
avrdude: load data application data from input file Chameleon-Mini.hex:
avrdude: input file Chameleon-Mini.hex contains 131072 bytes
avrdude: reading on-chip application data:

Reading | ################################################## | 100% 0.62s

avrdude: verifying ...
avrdude: 131072 bytes of application verified
avrdude: reading input file "Chameleon-Mini.eep"
avrdude: writing eeprom (164 bytes):

Writing | ################################################## | 100% 0.04s

avrdude: 164 bytes of eeprom written
avrdude: verifying eeprom memory against Chameleon-Mini.eep:
avrdude: load data eeprom data from input file Chameleon-Mini.eep:
avrdude: input file Chameleon-Mini.eep contains 164 bytes
avrdude: reading on-chip eeprom data:

Reading | ################################################## | 100% 0.01s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0002
         0xde != 0x5e
avrdude: verification error; content mismatch

avrdude done.  Thank you.

skuep, 2017-11-02:
Thanks! First thing that comes to mind is, that you are not erasing the eeprom before writing it. This might be the problem. Can you try to perform an eeprom erase before programming the eep file?

tned73, 2017-11-02:
Thanks, tried a couple of 
>sudo avrdude -c flip2 -p ATXMega128A4U -B 60 -P usb -e

But nothing, flashing still give the same error and restarting (unplug and replug) shows nothing dmesg.

No problem if the eeprom is damaged, I probably replace the Atmel, I just want to know what caused it. So can avoid it next time.

This is the pseudo command sequence what has been looping:
 >   cmdLogClear(chameleon)
    cmdRedLED(chameleon, 'FIELD_DETECTED')
    cmdUpload(chameleon, image)
    cmdConfig(chameleon, imagetype)
-- Wait for finish test
    cmdConfig(chameleon, "NONE")    
    cmdLogMode("OFF")            
    cmdRedLED(chameleon, 'NONE')
    cmdDownload(chameleon,len(image))
    cmdLog(chameleon)

tned73, 2017-11-02:
Thanks for the input, I will read through the code and see where eeprom writing happens and maybe try to implement a option to disable it.

Thanks for the input.

PS any suggestion where I can find new xmega128a4u-u in TQFP package type or compatible in Europe.

tned73, 2017-11-02:
The Atmel datasheets tells me that the eeprom should at least do a 100K Write/Erase cycles. yep, probably surpassed that. 
Atmel-8387-8-and16-bit-AVR-Microcontroller-XMEGA-A4U_Datasheet.pdf

tned73, 2017-11-02:
Or could I save the Atmel's by moving the eeprom starting point in the firmware? from the flashing log it seems we only need 164 bytes and we have 2k available? Eeprom verification will probably fail, but that I can live with.

tned73, 2017-11-03:
I found a makefile option
ENABLE_EEPROM_SETTINGS
Which for my use I should probably disable .
now searching how to move EEPROM startpoint

tned73, 2017-11-03:
I think I am able to avoid EEPROM use at all. I will close this Issue and probably will supply a pull request in the near future, to solve the next issues
Issues found if eeprom settings disabled: default settings still read from EEPROM and a Log setting is during init read and set in eeprom.

emsec, 2017-11-03:
I don' get why a "disabling the EEPROM" is necessary in your automatized application. I would rather execute the configuration commands such as config=mf_classic_1k and set the LED etc. (everything that touches the EEPROM) only ONCE during initialization and then repeatedly execute only the upload (and test) command. 

More precisely: why do you loop 

    cmdLogClear(chameleon)
    cmdRedLED(chameleon, 'FIELD_DETECTED')
    cmdUpload(chameleon, image)
    cmdConfig(chameleon, imagetype)
    -- Wait for finish test
    cmdConfig(chameleon, "NONE")
    cmdLogMode("OFF")
    cmdRedLED(chameleon, 'NONE')
    cmdDownload(chameleon,len(image))
    cmdLog(chameleon)

when you could EXECUTE ONCE:
    cmdLogClear(chameleon)
    cmdRedLED(chameleon, 'FIELD_DETECTED')
    cmdConfig(chameleon, imagetype)

and then 500.000 times LOOP:
    cmdUpload(chameleon, image)
    -- Wait for finish test
    download and clear log
(END LOOP)

?? 

Is there s.th. I didn't understand?

I think all the following commands make no sense:
    cmdConfig(chameleon, "NONE")
    cmdLogMode("OFF")
    cmdRedLED(chameleon, 'NONE')
    cmdDownload(chameleon,len(image))
    cmdLog(chameleon)

Cheers
Timo

tned73, 2017-11-03:
Dear Timo,

Thanks for your interest in my problem. 
As I see it, the eeprom functionality to keep the settings over a power cycle is handy but not needed for my needs. My use is to feed different types of images to the Chameleon-Mini and let the reader interact with the image. Continuously powered by USB from a Raspberry Pi. There is no need for me to keep settings over a power cycle.

The commands that make no sens to you:
Led, have a visual so I can see that the reader should interact
Set Conifg to NONE, make sure that the reader does not interact with the image anymore
Download is to get that particular image back in the size it was written, to evaluate what has changed
and the log to see how the communication has run.

I am not sure that a upload of a new image, would be noticed by the application and reset it, else it would stick in the Halted situation. And the new image would not be detected. especially if the Chameleon keeps being in the NFC field. Even if it does, it should only interact once the image is complete.

If I knew the enable_eeprom option sooner I would have disabled it, and it would not have ruined the eeprom. 
So for me was the easy step to extend the already available option to not use the eeprom and make it a bit more in a way that it really doesn't use the eeprom, only if you enable it. Mainly to work around my ruined eeprom's. Which was a lot quicker then to desolder the xmega's

I Hope I explained my reasoning a bit clearer.

Cheers.

closed 2017-11-09 by tned73