emsec/ChameleonMini #153: Opportunity to generate masked UID only in middle random?

gtpy, 2017-11-17:
Is it possible to generate for BUTTON_ACTION like UID_RANDOM
a functionality to generate a masked UID where only the middle is random?

Example UID for mf_classic_4k_7b (**04**_096E023F_**2280**):
begins only with "04"
ends every time with "2280"

same functionality that it begins with 04 and ends every time with 2480 - middle is random.

I am a really beginner in c - will be great if someone has an idea how this could be possible.

geo-rg, 2017-11-20:
This is possible, i suggest you have a look in `Button.c`. Right after `case BUTTON_ACTION_UID_RANDOM:` you can find a for loop that starts from 0 and goes up to UidSize. Right before the loop you will need to call `ApplicationGetUid` to obtain the currently set UID and you need to let the loop run from 1 to <UidSize-2. That should do it.

gtpy, 2017-11-20:
Thanks @geo-rg 

But for my better understandig:

`case BUTTON_ACTION_UID_RANDOM:`
`    {`
`        ApplicationGetUid(UidBuffer);`
`        for (uint8_t i=1; i<ActiveConfiguration.UidSize-2; i++) {`
`            UidBuffer[i] = RandomGetByte();`
`        }`

`        ApplicationSetUid(UidBuffer);`
`break;`

Would this be the right way?

Thanls a lot.

gtpy, 2017-11-20:
Thanks. Tried it out.
Works well. Thanks @geo-rg

closed 2017-11-20 by gtpy

herrmanns, 2019-01-27:
can anybody send me the chameleon firmware with this "button" option. please i am somehow not able to get this to work.

gtpy, 2019-01-27:
For The masked uid I can send you the fw. Or I can assist you how to change the code. At Wednesday I am back from holiday

-------- Original-Nachricht --------

herrmanns, 2019-01-27:
that sounds great gtpy. shall we use some email account or irc or anything (i dont have any social media things like fb, instagram etc).
i dont want to spam github. btw i am from GER, too.

gtpy, 2019-01-29:
@herrmanns How can I contact you?

gtpy, 2019-01-31:
You got a mail ;)