emsec/ChameleonMini #154: Error on flashing firmware (Win 10)

BlackBooth, 2017-11-18:
Hi, I got my Chameleon Mini today (yes, an original one) and followed the "Getting Started Guide". Flashing the firmware ends with the following error:

```
Checking memory from 0x0 to 0x1FFFF...  Not blank at 0x1.
Erasing flash...  Success
Checking memory from 0x0 to 0x1FFFF...  Empty.
0%                            100%  Programming 0x60 bytes...
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
0%                            100%  Reading 0x800 bytes...
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
Validating...  Success
0x60 bytes written into 0x800 bytes memory (4.69%).
Checking memory from 0x0 to 0x1FFFF...  Empty.
0%                            100%  Programming 0x20000 bytes...
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
0%                            100%  Reading 0x20000 bytes...
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
Validating...  ERROR
256 invalid bytes in program region, 0 outside region.
FAIL
Memory did not validate. Did you erase?
There was an error with executing this command. Maybe your ChameleonMini is not in bootloader mode?
```

The Chameleon is in bootloader mode. OS is Windows 10, latest version. Flashing the firmware with FLIP fails on verifiying with error `Verify device fails at 0x13C00`.

Any ideas? Thanks in advance

geo-rg, 2017-11-20:
Hi @BlackBooth 
have you, before trying to flash it, started the Chameleon (green LED turned on) and connected to it via terminal (test firmware with `showlog` command etc.)? Just to make sure if it has worked before...

BlackBooth, 2017-11-20:
Hi @geo-rg 
yes the existing firmware works fine. Connectiong with terminal and chameleonminigui also works.
Here is the output of the `version?` command:
```
ChameleonMini RevG 171107 using LUFA 151115 compiled with AVR-GCC 4.9.2. Based on the open-source NFC tool ChameleonMini. https://github.com/emsec/ChameleonMini commit 33f3525
```

geo-rg, 2017-11-20:
@BlackBooth Now that's a surprise. This firmware is that one you have flashed. The test firmware which is pre-programmed before delivery would show different data. 

Have you tried flashing multiple times? Did flip always respond with the same address where the verify failed?

BlackBooth, 2017-11-20:
@geo-rg Yes I tried flashing multiple times. I have to check the messages from FLIP when I'm back home.

BlackBooth, 2017-11-20:
@geo-rg I can confirm that FLIP always responds with the same address `Verify device fails at 0x13C00`. But still, the firmware seem to work, although veryfing does not succeed.

skuep, 2017-11-20:
0x13C00 sounds suspiciously like the start of the bootloader. It actually shouldn't check it, as it should not be within the hex file.

geo-rg, 2017-11-20:
@skuep The start of bootloader is 0x10000 as word address, but it ends at 0x10FFF (word address). In bytes this is from 0x20000 to 0x21FFF. Thus, it has nothing to do with the bootloader.

If 0x13C00 is a byte address, it is within the 2. slot/setting content and because of that it does not disturb the firmware. @BlackBooth Tomorrow I will post some instructions to test whether there is a problem with the flash at this position. Basically, we need to upload a dump to slot 2, store it to flash, then recall it, download it back again and compare whether uploaded and downloaded data are the same.

skuep, 2017-11-20:
Georg, thanks for checking!

geo-rg, 2017-11-21:
@BlackBooth Please download this random dump:
[8KB_rand.mfd.zip](https://github.com/emsec/ChameleonMini/files/1490157/8KB_rand.mfd.zip)

Then follow these steps:
1. Boot your Chameleon
2. Type `setting=2` in the terminal
3. Type `upload` in the terminal
4. upload the .mfd file with xterm
5. `store`
6. `recall`
7. `download` and download to another file via xterm

Then compare the uploaded and the downloaded file. Do they differ? If yes, the flash memory of your uC might be broken (or, that one page of the flash).

If you have trouble with any of these instructions (e.g., uploading/downloading dumps), don't hesitate to write here.  If you have problems with comparing two binary files, you could also simply put the downloaded dump into a zip/rar archive and upload it here so I can do the comparison.

BlackBooth, 2017-11-21:
@geo-rg You are right, the downloaded file is different from the uploaded file:
```
* MD5: 2dcc97d1cf9013a8114bca8093d824f3
* MD5: 0a4c64edb2deee74a442b81bc67114f8
* SHA1: cc5829c2a6c3b2136162c101a8c144ba7ca6ea89
* SHA1: bcc6c004327d75ca3ffead794995a1c7e33d09cc
```
Screenshot of WinMerge:
![Uploading 2017-11-21 10_35_32-WinMerge - [8KB_rand.mfd - 8KB_randdown.mfd].png…]()

Looks like I need a new one :/

geo-rg, 2017-11-21:
Hi @BlackBooth,
please write an E-Mail to chameleon@kasper-oswald.de in order to get a
replacement of your Chameleon.

BlackBooth, 2017-11-21:
Thank you, I will do that!

skuep, 2017-11-21:
I am curious, is there anything in particular that you did with the ChameleonMini to wear out the second slot? You are the second person to experience this. Maybe there is an easy fix in software to prevent others breaking their flash too.

BlackBooth, 2017-11-22:
@skuep Not that I'm aware of. As soon as I got my Chameleon I followed the Getting Started guide and tried to flash the firmware. That's where the error occured.

closed 2017-12-01 by skuep