emsec/ChameleonMini #155: How are you editing your Git data/projects?

gtpy, 2017-11-22:
How are you editing your Git data?
Are you using any IDE and import it?

Want to try some contributions but not really often worked with GitHub.

Whats the best way - I tried out Eclipse - but got a lot of "could not be resolved" and some syntax errors.

Thanks.

geo-rg, 2017-11-22:
I am using Eclipse. That "could not be resolved" is a problem of the indexer. Do you have the AVR plugin for Eclipse? If not so, you should install it. Then open Project->Properties. Is there a AVR tab on the left? If so, you can change the target hardware there to atxmega128a4u and rebuild the index. If not so, you can change the target hardware within C/C++ Build -> Environment. Change the value of AVRTARGETMCU to atxmega128a4u. It is possible that you will also have to change something in C/C++ General -> Paths and Symbols.

Most likely installing the AVR plugin should be sufficient.

gtpy, 2017-11-22:
I changed the target hardware to atxmega128a4u in C/C++ Build -> Environment. And changed in C/C++ General -> Paths and Symbols __AVR_DEVICE_NAME__ = atxmega128a4u
But now (I imported it as file -> new -> makefile project with existing code - it shows less messages "could not be resolved.
For example:
`Field 'BUTTON_L_PINCTRL' could not be resolved	Button.c	/Chameleon-Mini	Zeile 195	Semantic Error`
Most errors disappear.

geo-rg, 2017-11-23:
@gtpy Which errors do not disappear?

gtpy, 2017-11-23:
@geo-rg 

I am not sure which other Symbol value could not be correct.

...
Type 'USB_Descriptor_String_t' could not be resolved	LUFADescriptors.c	/Chameleon-Mini	Zeile 193	Semantic Error
Type 'USB_Descriptor_String_t' could not be resolved	LUFADescriptors.c	/Chameleon-Mini	Zeile 204	Semantic Error
Type 'USB_Descriptor_String_t' could not be resolved	LUFADescriptors.c	/Chameleon-Mini	Zeile 215	Semantic Error
Type 'USB_Descriptor_Interface_t' could not be resolved	LUFADescriptors.c	/Chameleon-Mini	Zeile 99	Semantic Error
Type 'USB_Descriptor_Interface_t' could not be resolved	LUFADescriptors.c	/Chameleon-Mini	Zeile 154	Semantic Error
Type 'USB_Descriptor_Endpoint_t' could not be resolved	LUFADescriptors.c	/Chameleon-Mini	Zeile 144	Semantic Error
Type 'USB_Descriptor_Endpoint_t' could not be resolved	LUFADescriptors.c	/Chameleon-Mini	Zeile 170	Semantic Error
Type 'USB_Descriptor_Endpoint_t' could not be resolved	LUFADescriptors.c	/Chameleon-Mini	Zeile 180	Semantic Error
Type 'USB_Descriptor_Device_t' could not be resolved	LUFADescriptors.c	/Chameleon-Mini	Zeile 46	Semantic Error
Type 'USB_Descriptor_Device_t' could not be resolved	LUFADescriptors.c	/Chameleon-Mini	Zeile 48	Semantic Error
Type 'USB_Descriptor_Device_t' could not be resolved	LUFADescriptors.c	/Chameleon-Mini	Zeile 242	Semantic Error
Type 'USB_Descriptor_Configuration_Header_t' could not be resolved	LUFADescriptors.c	/Chameleon-Mini	Zeile 84	Semantic Error
Type 'USB_CDC_Descriptor_FunctionalUnion_t' could not be resolved	LUFADescriptors.c	/Chameleon-Mini	Zeile 135	Semantic Error
Type 'USB_CDC_Descriptor_FunctionalHeader_t' could not be resolved	LUFADescriptors.c	/Chameleon-Mini	Zeile 115	Semantic Error
Type 'USB_CDC_Descriptor_FunctionalACM_t' could not be resolved	LUFADescriptors.c	/Chameleon-Mini	Zeile 127	Semantic Error
Type 'pgm_read_byte(offsetof(NVM_PROD_SIGNATURES_t, USBRCOSCA))' could not be resolved	System.c	/Chameleon-Mini	Zeile 111	Semantic Error
Type 'pgm_read_byte(offsetof(NVM_PROD_SIGNATURES_t, USBRCOSC))' could not be resolved	System.c	/Chameleon-Mini	Zeile 112	Semantic Error
Symbol 'WDT' could not be resolved	System.c	/Chameleon-Mini	Zeile 35	Semantic Error
Symbol 'WDT' could not be resolved	System.c	/Chameleon-Mini	Zeile 84	Semantic Error
Symbol 'WDT_PER_512CLK_gc' could not be resolved	System.c	/Chameleon-Mini	Zeile 84	Semantic Error
Symbol 'WDT_ENABLE_bm' could not be resolved	System.c	/Chameleon-Mini	Zeile 84	Semantic Error
Symbol 'WDT_CEN_bm' could not be resolved	System.c	/Chameleon-Mini	Zeile 35	Semantic Error
Symbol 'WDT_CEN_bm' could not be resolved	System.c	/Chameleon-Mini	Zeile 84	Semantic Error
Symbol 'USE_INTERNAL_SERIAL' could not be resolved	LUFADescriptors.c	/Chameleon-Mini	Zeile 70	Semantic Error
Symbol 'USB_CONFIG_ATTR_SELFPOWERED' could not be resolved	LUFADescriptors.c	/Chameleon-Mini	Zeile 92	Semantic Error
Symbol 'USB_CONFIG_ATTR_RESERVED' could not be resolved	LUFADescriptors.c	/Chameleon-Mini	Zeile 92	Semantic Error
Symbol 'USARTD0' could not be resolved	Memory.c	/Chameleon-Mini	Zeile 43	Semantic Error
Symbol 'USARTD0' could not be resolved	Memory.c	/Chameleon-Mini	Zeile 45	Semantic Error
Symbol 'USARTD0' could not be resolved	Memory.c	/Chameleon-Mini	Zeile 47	Semantic Error
Symbol 'USARTD0' could not be resolved	Memory.c	/Chameleon-Mini	Zeile 343	Semantic Error
Symbol 'USARTD0' could not be resolved	Memory.c	/Chameleon-Mini	Zeile 344	Semantic Error
Symbol 'USARTD0' could not be resolved	Memory.c	/Chameleon-Mini	Zeile 345	Semantic Error
Symbol 'USARTD0' could not be resolved	Memory.c	/Chameleon-Mini	Zeile 346	Semantic Error
Symbol 'USARTD0' could not be resolved	Memory.c	/Chameleon-Mini	Zeile 352	Semantic Error
Symbol 'USARTD0' could not be resolved	Memory.c	/Chameleon-Mini	Zeile 353	Semantic Error
Symbol 'USARTD0' could not be resolved	Memory.c	/Chameleon-Mini	Zeile 366	Semantic Error
Symbol 'USARTD0' could not be resolved	Memory.c	/Chameleon-Mini	Zeile 367	Semantic Error
Symbol 'USART_TXEN_bm' could not be resolved	Memory.c	/Chameleon-Mini	Zeile 346	Semantic Error
Symbol 'USART_RXEN_bm' could not be resolved	Memory.c	/Chameleon-Mini	Zeile 346	Semantic Error
Symbol 'USART_RXCIF_bm' could not be resolved	Memory.c	/Chameleon-Mini	Zeile 45	Semantic Error
Symbol 'USART_CMODE_MSPI_gc' could not be resolved	Memory.c	/Chameleon-Mini	Zeile 345	Semantic Error
Symbol 'RTC' could not be resolved	System.c	/Chameleon-Mini	Zeile 61	Semantic Error
Symbol 'RTC' could not be resolved	System.c	/Chameleon-Mini	Zeile 62	Semantic Error
Symbol 'RTC' could not be resolved	System.c	/Chameleon-Mini	Zeile 63	Semantic Error
Symbol 'RTC' could not be resolved	System.c	/Chameleon-Mini	Zeile 64	Semantic Error
Symbol 'RTC_PRESCALER_DIV1_gc' could not be resolved	System.c	/Chameleon-Mini	Zeile 63	Semantic Error
Symbol 'RTC_OVFINTLVL_LO_gc' could not be resolved	System.c	/Chameleon-Mini	Zeile 64	Semantic Error
Symbol 'RST' could not be resolved	System.c	/Chameleon-Mini	Zeile 29	Semantic Error
Symbol 'RST' could not be resolved	System.c	/Chameleon-Mini	Zeile 32	Semantic Error
Symbol 'RST' could not be resolved	System.c	/Chameleon-Mini	Zeile 77	Semantic Error
Symbol 'RST_WDRF_bm' could not be resolved	System.c	/Chameleon-Mini	Zeile 29	Semantic Error
Symbol 'RST_WDRF_bm' could not be resolved	System.c	/Chameleon-Mini	Zeile 32	Semantic Error
Symbol 'RST_SWRST_bm' could not be resolved	System.c	/Chameleon-Mini	Zeile 77	Semantic Error
Symbol 'PORT_OPC_PULLUP_gc' could not be resolved	Button.c	/Chameleon-Mini	Zeile 194	Semantic Error
[... cut ...]
Symbol 'PIN3_bm' could not be resolved	Memory.c	/Chameleon-Mini	Zeile 341	Semantic Error
Symbol 'PIN1_bm' could not be resolved	Memory.c	/Chameleon-Mini	Zeile 338	Semantic Error
Symbol 'PIN1_bm' could not be resolved	Memory.c	/Chameleon-Mini	Zeile 341	Semantic Error
Symbol 'OSC' could not be resolved	System.c	/Chameleon-Mini	Zeile 41	Semantic Error
Symbol 'OSC' could not be resolved	System.c	/Chameleon-Mini	Zeile 42	Semantic Error
Symbol 'OSC' could not be resolved	System.c	/Chameleon-Mini	Zeile 43	Semantic Error
...

gtpy, 2018-06-26:
@geo-rg which eclipse version are you running?

geo-rg, 2018-06-27:
@gtpy I'm using Eclipse 4.7.1a

gtpy, 2018-06-27:
@geo-rg thanks.

Now I figured out what was the problem at my machine.

(For long time edited only with notepad++, and other IDE :-)

I use 4.7.3a and get it working (solving the “could not be resolved” problem).

Reinstalled eclipse java, added the cdt packages, added avr-plugin, Atmel toolchain, WinAVR, edited the AVRTARGETMCU and correct the include directories.

That was the way to solve it for me.

closed 2018-06-27 by gtpy