emsec/ChameleonMini #156: Since revE is discontinued (third-party clone)

iceman1001, 2017-11-24:
I finally got the source code for the third-party,  revE,     and will make a iceman fork of it.
Read more here:
http://www.proxmark.org/forum/viewtopic.php?pid=30365#p30365

*not an issue*

iceman1001, 2017-11-26:
..and its up!

https://github.com/iceman1001/ChameleonMini-rebooted

geo-rg, 2017-11-28:
I added both links to the [external contributions](https://github.com/emsec/ChameleonMini/wiki/External-Contributions) since this does not really belong here. Thanks for your contributions!

closed 2017-11-28 by geo-rg

iceman1001, 2017-11-28:
Thanks,  its definitly not an issue,  it was used to get the message out to the ppl who bought the third-part device that there is a source code available now.  

So much code here to analyse :)

Thank you for your hard and devote work, @geo-rg