emsec/ChameleonMini #157: How to emulate Mifare Ultralight C?

vicentecph, 2017-12-11:
I saw that Ultralight C can be emulated here https://www.kickstarter.com/projects/1980078555/chameleonmini-a-versatile-nfc-card-emulator-and-mo

but now after executing the config command on my laptop I saw that the only available options are 
NONE,MF_ULTRALIGHT,MF_CLASSIC_1K,MF_CLASSIC_1K_7B,MF_CLASSIC_4K,MF_CLASSIC_4K_7B,ISO14443A_SNIFF,ISO14443A_READER

Any help on how can I turn my Chameleon Mini into an Ultralight C tag?

Thanks.

geo-rg, 2017-12-14:
Hi @vicentecph 
the text in the link states "can emulate in _principle_". This means as soon as someone implemented it, which is not yet the case. The hardware, however, supports emulating it. Feel free to hand in any patches!

dev-zzo, 2017-12-18:
@vicentecph Is there anything specific to UL C you need to emulate (something breaks if you give it a plain UL card etc)?

digitalhoax, 2019-11-11:
Has this been solved by https://github.com/emsec/ChameleonMini/commit/96ff627f48382f65fdc6a7f3036ce471cdd4f8e5 ?

fptrs, 2019-11-11:
Hi @digitalhoax,
yes we added the pull request but until now I forgot to update the firmware files.

closed 2019-11-11 by fptrs

wowfoot, 2022-10-21:
Is Mifare Ultralight C supported?
UltralightC codes are blocked by "#if 0" in Firmware/Chameleon-Mini/Application/Reader14443A.c.
When "#if 0" is changed to "#if 1", Is Mifare Ultralight C supported?