emsec/ChameleonMini #158: no device present

peseta3, 2017-12-14:
Hallo together,
I am a Newbee,  I have the follow problem.
I make all I have to do with the getting started page

[http://rawgit.com/emsec/ChameleonMini/master/Doc/Doxygen/html/Page_GettingStarted.html](url)

And I get with win10 the Massage

> dfu-programmer: no device present.
> There was an error with executing this command. Maybe your ChameleonMini is not in bootloader mode?
> 
what can I do with this?

gtpy, 2017-12-14:
@peseta3 Is the LED at the chameleon off if connected it via USB in bootloader mode? - Run as administrator?

>Make sure the Chameleon is switched off (power switch on the side is in the position OFF). Now, hold >down the right button (RBUTTON) and keep it pressed while connecting the Chameleon to a PC via a >USB cable.

peseta3, 2017-12-14:
Yes the LED is off.
I press the black right Bottum and make an usb conection the the cable, together.

Yes I run it with the admin.

gtpy, 2017-12-14:
@peseta3  
Question: is it an original from Kasper&Oswald (Rev.G) or chameleon revE rebooted like from iceman1001/ChameleonMini-rebooted
Because you opened one issue here and in iceman1001/ChameleonMini-rebooted
Black button sounds like the RevE Rebooted
If it is so, then please close this issue and let the other one open.

peseta3, 2017-12-14:
ah ok
Yes it is so as you say.
But I get this page from the buyer Homepage with link to link to link bla bla

Ok You sa I am not right here.
thank you for all fast, very fast post.

closed 2017-12-14 by peseta3

peseta3, 2017-12-14:
hallo again,
iceman says I have to go to you
hm
and now

iceman1001, 2017-12-14:
Well, I say I don't offer support.  That project just got the firmware to compile...
Basic operations should be covered in the wiki on this repo.

peseta3, 2017-12-14:
Hmm I don`t understand it, the wiki?

I can not open it.

 

Mist my english is not so well.

geo-rg, 2017-12-19:
Hi @peseta3 
since your Chameleon is not a genuine one from us but a remake, we cannot give support, especially for the hardware. Your problem looks like a hardware problem, or maybe these Chameleons have another way to enter the bootloader? Did you get any description of how to do firmware upgrades? Anyways, I'm afraid, we cannot give support for this.

peseta3, 2017-12-19:
hmm
but the Company write on her Page that this Page is right for Questions and so on
please have a lock for this
[https://lab401.com/collections/rfid-tools/products/chameleon-mini-reve-rebooted](url)
you can see it under 
Community / Usage / Tutorial Resources
[https://github.com/emsec/ChameleonMini/wiki](url)

What can you tell me about that?

geo-rg, 2017-12-19:
Again I'm afraid, but we do not have any knowledge about the product that you have bought. The vendor used this open source project as a base. The have changed many things on hardware and since we do not know anything about what they did exactly, we assume they may also have changed the bootloader software. Anyways, to be clear: **we cannot give support for products that we are not related to**.

gtpy, 2017-12-19:
Hi @peseta3 

 The one from Lab401 is maybe own developed from them and also different/modified in the coding_

I never owned a Rev.E neither original nor clone.
But I think if someone of the community have time and money to check it out, they’ll help and contribute with some tips and descriptions. 
I can not do it until the new year (maybe I’ll buy the same and test and try to contribute for it)

Maybe you have already seen, that we (some people from the community) are about to expand the repository in the iceman-fork -> like descriptions and some guidelines for this device (see your other issue iceman1001/ChameleonMini-rebooted#9
You are cordially invited to contribute to it, with informations or other help.

Have you already contacted the Lab401 team about you problem or did you get the driver installed and the device running? (issue iceman1001/ChameleonMini-rebooted#9)

peseta3, 2017-12-19:
Hi geo-rg
but why can Lab401 link it to emsec when nobody knows about that????

peseta3, 2017-12-19:
Hallo gtpy,

The company Lab 401 is not the fast one with reanswere for my problems.
I can conect it with win10 with thair GUI-Software.
But in the moment I cannot test it in realtime.
I write them with a discription for firmeware update, but I didn`t heard something from them.

iceman1001, 2017-12-19:
Since the Lab401 revE rebooted version is based upon offical revE,  but it had a locked down firmware and a custom GUI.    As mentioned several times before,  I got the source code for the lockdown firmware and made it open-source again and some people are trying to figure out what it does and not.  
..and how to fix the obivous bugs and add all fixes from this repo..

All we got at the moment is the idea that the normal setting up / flashing instructions from this offical repo should work.  That is why everyone pointing back to this wiki here.   Noone knows, not the retailer (lab401) and the manufacturer just say "use the gui".  As @gtpy @geo-rg  et al are saying,  since its not a offical chameleon mini revE,   this repo can't help you either.

The work with filling the wiki at iceman-fork, has begun.  Its not a fast job not do anyone claim to give support,  its a voluntered one.  I understand its frustrating thing to not be able to use your new gadget but this is how it is right now.

peseta3, 2017-12-19:
ok danke iceman

peseta3, 2017-12-19:
@iceman1001 
corespodenz with Lab401

•	You can always flash back to the firmware we provide
•	You can always flash to whichever firmware you want
You will need the AVRISP MKII hardware to do any flashing.
https://lab401.com/collections/flashing-tools/products/avrisp-mkii?variant=5004374835231

You would use the software AVR Studio to do your flashing. There are many tutorials online, and it is very easy to do.
A good reference point is here: http://www.atmel.com/webdoc/avrispmkii/avrispmkii.intro_connecting.html

peseta3, 2017-12-19:
@iceman1001 

I am a absolute Newbee for that.
I don´t know what I do, or I have to do.
Iceman, do you speak german?
maybe I can speak  per phone or so a few minutes if you want and have time.
thanks a lot

iceman1001, 2017-12-19:
Sorry, I have too little time left as it is.

peseta3, 2017-12-19:
@iceman1001 
hmm ok
thank you for all

gtpy, 2017-12-19:
@peseta3 how can I contact you? Mail? IRC? Signal?
We can write in German and I can try to help you with the first steps.

peseta3, 2017-12-19:
great
chameleonatmtrackdotorg

iceman1001, 2017-12-19:
Some great succes tonite,  the firmware compiles and we figured out how to flash the device with it. 
Finally bugfixes and enhancements are available!