emsec/ChameleonMini #159: New Android-based logger and parser utility

maxieds, 2018-01-02:
I wasn't sure where else to post this, so I figure this is the best recommended place for it. I have written a new utility for my Android phone which is an interactive NFC debugging tool using the LIVE logging mode for the Rev. g boards. It is available on the Play Store (unreleased) [here](https://play.google.com/store/apps/details?id=com.maxieds.chameleonminilivedebugger), and the working source is available on my GitHub page [here](https://github.com/maxieds/ChameleonMiniLiveDebugger). I'm still actively working on developing the project and could use some testers for it. Also, I'd like to test it out on other Chameleon revisions and boards (I bought an official one on Amazon some weeks back). Is there anyway I can get my hands on some older hardware to get support for it working / verified with the app? Any suggestions for future useful features that I haven't thought of are also welcome. Thanks. 

Maxie

gtpy, 2018-01-02:
@maxieds sounds great.
If I have some time I’ll take a look to it. I can contribute with testing with a Rev. G 
Cheers

maxieds, 2018-01-02:
Used one of [these](https://www.amazon.com/gp/product/B00CXAC1ZW/ref=oh_aui_detailpage_o03_s00?ie=UTF8&psc=1) USB cables to get the Rev. g board hooked up to my phone. It would be nice if the Chameleon boards were smaller like some of the compact wifi usb adapters. You still need to lug around a lot of parts to use the interactive debugging, but I guess it's better than being tethered to a laptop and live console to get NFC debugging data.

BTW, roughly 1/4 of the buttons on the Tools Menu tab are status markers so that if you move around testing different door readers you have ways to denote this in the resulting log files. The colors are for highlighting and most of the rest are commands or configure-select options built-in to the Chameleon command line.

gtpy, 2018-01-02:
Nice. Normally I use for a connection with smartphones a normal OTG adapter and serial usb terminal app. Your app  isn’t available to download from play store in Germany. I’ll load it via proxy and then test it on my development phones.
Greetings

maxieds, 2018-01-02:
I submitted a new revision of the app to the Play Store. It includes improved navigation in Tools Menu and writing features for writing the captured live logs to file. The binary file format is identical (up to timestamp discrepancies) the native internal logs produced by LOGDOWNLOAD, and so it can be parsed with the stock Python scripts shipped with this main repo.

I also added store support for Germany and the UK. These have to be enabled one-by-one, so those two were the most obvious to include after your suggestion. 

Enjoy!

maxieds, 2018-01-02:
One other question I have about the Chameleon device is whether it is possible to setup a configuration that is based on log replay. So suppose I do not have the full specs for a given card, but I know several of it's deterministic responses to queries from a reader. Can I setup a device that will respond with the known data from my logs at just these queries?

maxieds, 2018-01-04:
I added several new features including more command / tools buttons and tentative recognition of APDU commands in the data from the logs. A signed APK for the app will be on the Play Store soon (next couple of days). 

I could still use more testers and users to give me suggestions on features!

Recent Debugging APK: [ChameleonMiniLiveDebugger-0.0.5-v1.zip](https://github.com/emsec/ChameleonMini/files/1603377/ChameleonMiniLiveDebugger-0.0.5-v1.zip)

geo-rg, 2018-01-08:
Just added it to the [External Contributions page](https://github.com/emsec/ChameleonMini/wiki/External-Contributions). I did not have the time to test it, but the screen shots look awesome, thanks!

AndreasBujok, 2018-01-08:
Yes, but my Samsung S4 is to old told me the  Marketplace :-(

maxieds, 2018-01-08:
I'm always looking for new feature requests. I'm also curious as to how
accurate the current apdu command recognition is. Any one who has time to
test this, would you make some suggestions for improvement?

Re: Samsung S4 phone
Sorry it doesn't work out of the box with your phone. I needed some
features that made me push the API level up near 24 in Android Studio. Can
you upgrade the software on your phone to the lastest version? If not, you
could try modifying the source on the GitHub page to get it working with
your phone. My phone I'd from 2015, so I'm not exactly using bleeding edge
hardware either. Let me know if either method works.

Maxie

maxieds, 2018-01-08:
Also, I've noticed that if you put the Chameleon into reader mode and turn
on NFC on the Droid phone, the app will log transactions between the phone
and any tag / desfire card you have near it.

Personally I bought the chameleon so I could try to clone my university I'd
card with my phone (eventually without the chameleon), so it's definitely
useful for logging interactions between a desfire card, Android, and the
aptiQ line of door readers. Android just doesn't give you enough data about
the transactions that are really passing through it.

maxieds, 2018-01-08:
Should be "sniffer" mode from the app menu actually.

maxieds, 2018-01-09:
# This thread officially moved context to the main app repo issues site

**_FYI:_** Testing comments and feature requests can now be made on the main app project issues page at [https://github.com/maxieds/ChameleonMiniLiveDebugger/issues](https://github.com/maxieds/ChameleonMiniLiveDebugger/issues). [#2](https://github.com/maxieds/ChameleonMiniLiveDebugger/issues/2)

**Other Notes:**
_I created the app because I needed something that would do this in a little more user-friendly display way than communicating with the Chameleon Mini over the default serial terminal method outlined in the wiki docs. If you use it, like it, run into issues, or think is needs other features to be more useful to your usage and setup,_##_**PLEASE COMMENT AND LET ME KNOW!**_ ##

**Developer Requests:**
:moneybag: Also, the app is freeware on the playstore, so instead of adding a PayPal account for good Samaritan user donations, I will just point out that I would like to have another couple of these devices to play with and loan to friends for another private app I'm developing. If anyone finds this app useful and has the resources to donate an extra Chameleon or two (any revision should do the trick), please contact me at mygithubhandle@gmail.com. My math university webpage should also be linked off my GitHub profile, so you can contact me that way as well. The $155 Amazon price tag for duplicate Chameleon's is a little much for my budget right now. The free source code, however, is a labor of love and will continue to be supported to the best of my ability.:sparkling_heart:

:wavy_dash:I also would appreciate it if you would :star2: star the source repo [here](https://github.com/maxieds/ChameleonMiniLiveDebugger/stargazers) :star2: if you use the app or download it from the play store.:wavy_dash:

maxieds, 2018-01-09:
@AndreasBujok (RE: Your Samsung S4 Phone)

I found the following links which should help you upgrade the software on your older droid to something that will be usable with the app. I'm currently trying to get the feature set in the app paired down so it will work with Android 4.4 (KitKat), which I believe anyway is the first release where mainstream NFC card emulation support was introduced. I'm planning on rolling out a new Play Store release tonight, so you can check back then to see if the app will now work with your device when I'm done with my main round of edits for today. _Enjoy!_

- [https://android.stackexchange.com/questions/144918/how-to-update-galaxy-s4-to-lollipop-or-latest-android-version](https://android.stackexchange.com/questions/144918/how-to-update-galaxy-s4-to-lollipop-or-latest-android-version)
- [https://android.stackexchange.com/questions/56112/when-will-my-device-get-the-android-4-4-update-kitkat](https://android.stackexchange.com/questions/56112/when-will-my-device-get-the-android-4-4-update-kitkat): See the Samsung S4 support by scrolling to midpage
- [https://android.stackexchange.com/questions/13510/how-do-i-update-android-on-my-device](https://android.stackexchange.com/questions/13510/how-do-i-update-android-on-my-device)

Are you able to install from one of the rollouts in the countries I have added so far? Again, I've mentioned this before, but it's really a royal PITA that Google segregates it's play store targeting so much. They should just have an option to roll it out to everyone but the banned countries (I'm located in the US) on the developer's source export list and be done with it. * * sigh * *

maxieds, 2018-01-10:
A [new version](https://play.google.com/store/apps/details?id=com.maxieds.chameleonminilivedebugger) (probably the last one for a while) is available. I tried to make it more portable to more devices. Includes bug fixes and a number of new features. Please check it out and enjoy!

maxieds, 2018-01-13:
I have posted a new [release here](https://github.com/maxieds/ChameleonMiniLiveDebugger/releases) and it is now on [play store here](https://play.google.com/store/apps/details?id=com.maxieds.chameleonminilivedebugger) (v0.1.4). Includes many new features, improvements, and bug and stability fixes. Also, XModem downloads of logs and card data to sdcard files are now supported and functional. If you haven't tried it out yet, please do so! I think it's a much easier way to access and configure the Chameleon mini than with the standard serial console. Improvements and feature requests can be given [here](https://github.com/maxieds/ChameleonMiniLiveDebugger/issues). -- Maxie

maxieds, 2018-01-15:
### Wiki Link Request?
@geo-rg, @gtpy:
Is there any way I could get a link to the app on the [Getting Started](https://rawgit.com/emsec/ChameleonMini/master/Doc/Doxygen/html/Page_GettingStarted.html) page? I appreciate the link on the [External Contributions](https://github.com/emsec/ChameleonMini/wiki/External-Contributions), but if you have the USB cable adapter needed to hook the standard cable that ships with the Chameleon Mini to your phone, I really feel that this app is so much easier to use for the first time than trying to use it with a PC-based serial console. Not sure how picky you all are about referring external contributions on the main page... :pray: I have put a lot of effort into trying to make the app usable for Chameleon newbies like myself.

gtpy, 2018-01-15:
@maxieds i think it is a good idea to link your really nice work to the getting started page.
I’ll checkout the new version today or the next few days.
@geo-rg what are you thinking about this?

maxieds, 2018-01-15:
I am fixing a couple of remaining issues about how the app responds when
the USB device is not initially plugged in. I also added a slider bar for
setting the device threshold to the version currently rolled out. Please
wait for v0.1.9 to be posted on the Play Store today before you test it
out. This should be up / rolled out in the next few hours. Thanks for
including a link on the getting started page!

maxieds, 2018-01-15:
**Update:** The new version (v0.1.9) is now available on the [Play Store](https://play.google.com/store/apps/details?id=com.maxieds.chameleonminilivedebugger) and on the [app repo](https://github.com/maxieds/ChameleonMiniLiveDebugger/releases/tag/v0.1.9). Enjoy!

gtpy, 2018-01-17:
@geo-rg Is it possible to add a link to the doxygen “getting started” to refer to the android app as a mobile solution?
The actual version 0.2.6 is running stable and really simple to use. I already tested the functionalities.

emsec, 2018-01-17:
@gtpy I will do this as soon as possible, most likely today! Thanks for all your efforts @maxieds @gtpy

gtpy, 2018-01-17:
Thanks a lot.

Sent from ProtonMail Mobile

maxieds, 2018-01-17:
The latest version of the app is [v0.2.8](https://github.com/maxieds/ChameleonMiniLiveDebugger/releases/tag/v0.2.8). There are also now matching themes / color profiles on the repo site for download. The latest Play Store APK will remain in the default green color scheme ("main" flavor). If you use the app and like it, please rate the app on Play Store. I would like to grow the user base somewhat now that the project is stable.

geo-rg, 2018-01-17:
@maxieds @gtpy Just added it.

gtpy, 2018-01-17:
@geo-rg thanks a lot.

maxieds, 2018-01-20:
@geo-rg, @gtpy: Thanks for the links! Also, thanks to @doegox

maxieds, 2018-01-21:
### "Paid" flavor on Play Store now wraps up the themed UI features:
I decided to create a ["paid" flavor](https://play.google.com/store/apps/details?id=com.maxieds.chameleonminilivedebugger.paid) of the application for those who want to support the project, but do not have the resources to contribute pricey hardware to the development process ($5 for themes). An example "Redmond" (someone besides me should be laughing at this, :dollar: :euro:) theme is featured in the following screenshots for reference:

[image]
[image]

I might also add some testing Desfire card implementations in **res/raw** to the "paid" flavor at some point to incentivize more people who actually use the application to support the development process. Free license, not endless developer resources to contribute kind of thing. At any rate, enjoy the mock theme. The same new release (v0.2.9-free) is available [here](https://github.com/maxieds/ChameleonMiniLiveDebugger/releases/tag/v0.2.9) for testing.