emsec/ChameleonMini #160: Wikipedia entry for the Chameleon Mini marked for speedy deletion

maxieds, 2018-01-05:
Greetings. While searching around for more information on the Chameleon Mini, I noticed that Wikipedia was missing an entry on this awesome device! I created the corresponding [entry](https://en.wikipedia.org/wiki/Chameleon_mini_(hardware)) and it was flagged for speedy deletion for it seeming to be promotional of a specific product or manufacturer. I have updated the original listing to make it "appear more encyclopedic". Would someone please visit that page and remove the flag for speedy deletion. For obvious reasons I don't want to have to be the one that does this. While you're there feel free to correct my descriptions of the devices and add anything you think is pertinent to the listing. Thanks.

ceres-c, 2018-01-05:
Too late, it seems to be already gone...

jtl21, 2018-01-05:
Too bad it was really well done! I hope we can get it back up.

geo-rg, 2018-01-08:
@maxieds Thanks for your efforts! Wikipedia often is very sensitive regarding potentially promotional contents and I also think that with a relatively small community of users such as ours, maybe the Wikipedia community also would raise objections regarding the relevance. As much as I would like an entry (and in my opinion, it cannot be promotional for a specific manufacturer, since it's open source), I think such an entry currently is not the most important thing for the project :) 

Nonetheless: thanks for your efforts!

maxieds, 2018-01-10:
I'm not going to argue with the *$$ of an editor who flagged it for deletion, obviously got annoyed when I insulted his intelligence *ahem* knowledge of the topic in question, and then deleted it anyway. I will however post the source to the page I created (careful: some of the wording is from the Amazon listing) if someone else wants to try it again. If it gets flagged for speedy deletion again, I check my mail often enough that another post to this thread should get me to remove the flag for you. 

**_## [wiki-cmhardware-source-page-only.txt](https://github.com/emsec/ChameleonMini/files/1617398/wiki-cmhardware-source-page-only.txt)_**

**BTW:** Here is the talk page exchange between myself and the editor. I'm obviously being facetious in places and adding sarcasm where appropriate to react to this (apparently notably rude) editor's flagging of the page:

###== Contested deletion ==
This page is not unambiguously promotional, because... (your reason here) --[[User:Maxieds|Maxie]] ([[User talk:Maxieds|talk]]) 22:48, 4 January 2018 (UTC)

This is a serious hardware tool for security analysis of the NFC protocol. A Wikipedia entry was missing despite being the first page of entries on google when searching for "Chameleon mini" (there is another popular Android application called _mini Chameleon_ with some following), so I added it. I will work on rewriting the article to make it more encyclopedic promptly. Please do not delete it.

###== Contested deletion ==

This page is not unambiguously promotional, because... (your reason here) --[[User:Maxieds|Maxie]] ([[User talk:Maxieds|talk]]) 23:28, 4 January 2018 (UTC)

This is a serious security tool in hardware form that was (previously) lacking a Wikipedia entry. Thus it was added upon this realization.

###==[[User:Boleyn|@Boleyn]]: Updated the page you marked for speedy deletion to appear more encyclopedic like it was originally intended==

Please see [[Chameleon_mini_(hardware)]] and the attached reasonable [https://en.wikipedia.org/wiki/Talk:Chameleon_mini_(hardware)| contest] to its speedy deletion. I would like for '''you''' to remove the banner, not for me to have to do it and then have an editor flag it again because the edit is found disagreeable to their tastes and/or knowledge of the underlying topic matter. It's a serious NFC security tool (that hackers, feds, terrorists et. al. all know about and use) that has many uses. The Wikipedia page was missing, so I added it. The underlying "OS" (if you will) driving this piece of hardware, that *'''yes'''* like most things, is sold commercially is a well-respected suite of open-source software with many followers. Additionally, the schematics for reproducing the device are freely available to the public under an OSS licence and consequently multiple manufacturers have produced versions of the device for commercial retail applications (the Rev. E and G boards have ''different'' manufacturers -- no touting of a single product here).  <small class="autosigned">—&nbsp;Preceding [[Wikipedia:Signatures|unsigned]] comment added by [[User:Maxieds|Maxieds]] ([[User talk:Maxieds#top|talk]] • [[Special:Contributions/Maxieds|contribs]]) 23:25, 4 January 2018 (UTC)</small>