emsec/ChameleonMini #161: Open Pull Requests - no merge timeline

rafb0y, 2018-01-08:
Hi, I just supported your project and bought a ChameleonMini Rev.G.
Since I am more interested in the rfid communication protocol than in software development, i realized that a lot of missing features have already been implemented by the community but not merged yet.
I would be interested in a compiled hex file :-).
Could you give us a timeline?
Thank you.

geo-rg, 2018-01-08:
First of all, we appreciate all the contributions! The main problem is that many things are not tested and the main branch shall contain tested code. Also, there are pull requests that are not yet finished, such as the ISO 15693 codec.

Which functionality do you desire in particular?

rafb0y, 2018-01-08:
https://github.com/emsec/ChameleonMini/pull/118

geo-rg, 2018-01-24:
Just merged it, but it is not yet tested and the compiled "Latest" version is not yet updated. It would be great if you could check out the repo, compile the code and test if everything works fine.

closed 2018-01-24 by geo-rg

geo-rg, 2018-01-25:
@rafb0y I also updated the "latest" precompiled version right now.