emsec/ChameleonMini #164: How can i rebuild the ChameleonMini with own changes

AndreasBujok, 2018-01-24:
Hi,

i want to modify/adapt some files and than recompile to get the new firmware files.
What do i need and what are the needed steps?

Thanks.

Cheers Andreas

geo-rg, 2018-01-24:
Hi @AndreasBujok 
to set up the toolchain, [this](http://avr-eclipse.sourceforge.net/wiki/index.php/The_AVR_GCC_Toolchain) is a good start. You then only need to navigate to your local ChameleonMini/Firmware/Chameleon-Mini directory in a terminal and run `make`. This will generate both the hex and the eep file this directory.

I suggest using linux since you will have set up everything after running one or two commands.

I also suggest leaving this issue open until you have succeeded compiling. If you have further problems compiling the code, you can also post them in this issue.

AndreasBujok, 2018-01-24:
Great, thanks for the support.

If a setup a ubuntu VM, is this a good base for it?

geo-rg, 2018-01-24:
Yes, this should be also ok. I really would recommend it ;)

AndreasBujok, 2018-01-26:
Hi @geo-rg 

Sounds good and easy. 
If a making a "wrong" firmware, i'm able to bring the ChameleonMini back to live by upload a previous firmware?

geo-rg, 2018-01-26:
@AndreasBujok Yes, as long as you don't do anything that could overwrite the bootloader section, which would include using functions that write to the flash memory (however, as long as you use MemoryStore() or similar functions, this should not happen). Another way to possibly harm the bootloader could be to use more or less push than pop in Assembly code. I'm just writing this to sensitize you that something _could_ happen although it is very unlikely that you cause something like this.

Even when you would harm the bootloader (that piece of code that jumps either to your application software or, if you hold the right button while powering the device, initiates the firmware upgrade procedure), you can restore this, but this is not possible via USB. In order to do this, you would need a so-called avr dragon.

AndreasBujok, 2018-01-26:
OK, fine.

Now, let's go modify ;-)
If i got a ChameleonMini Brick i will send to you to reanimate _*fg*_

Thanks a lot

AndreasBujok, 2018-01-27:
Hi @geo-rg.

Works all fine. I've also a step by step HowTo on my documentation https://github.com/AndreasBujok/ChameleonMini

Thanks. :-)

closed 2018-01-27 by AndreasBujok