emsec/ChameleonMini #165: clone mode improvements

geo-rg, 2018-01-25:
As you may have seen, the latest commits also contain a new clone mode which has been implemented by @robots - thanks for that! The related pull request was #118.

However, there are still some things to be done:
1. **Visual feedback** in case of clone success. I suggest adding an LED function (called "CLONE_DONE" or anything like that) which blinks a few times a soon as the clone process is finished.
2. **Documentation**. A documentation about how to use the clone mode should be added to the existing doxygen based documentation (see Doc folder).
3. **Automatic read of Mifare Ultralight**. Right now, only the UIDs are cloned. But for the unencrypted Mifare Ultralight, it would be possible to read its contents and afterwards emulate this type of card.

It would be great if someone could implement/write these things and hand in pull requests or patches (also possible via E-Mail to chameleon@kasper-oswald.de, although this would not be our preferred way).

robots, 2018-01-25:
I have nr. 3 as WIP. I will finish it now that i see there is interest. Nr. 1 can be added as well. But ill leave the documentation to someone else :-)

AndreasBujok, 2018-01-26:
@geo-rg @rafb0y 
Nice work 👍 
I've updated my docu https://github.com/AndreasBujok/ChameleonMini#clone

hsanjuan, 2018-12-27:
@robots did you ever submit n. 3 ?