emsec/ChameleonMini #166: sniffer mode clarification

hydra228, 2018-02-02:
I send the apdu command to my paypass mastercard, using the arduino + pn532 bundle.
in order of action:
0x00, 0xA4, 0x04, 0x00, 0x0E, 0x32, 0x50, 0x41, 0x59, 0x2E, 0x53, 0x59, 0x53, 0x2E, 0x44, 0x44, 0x46, 0x30, 0x31, 0x00
0x00, 0xA4, 0x04, 0x00, 0x07, 0xA0, 0x00, 0x00, 0x00, 0x04, 0x10, 0x10, 0x00
0x80, 0xA8, 0x00, 0x00, 0x02, 0x83, 0x00, 0x00

I get a log:

`## USB STATUS: : Successfully configured the device in passive logging mode. (@2018-02-02-00:44:34)
000001 -- LOG_CODE_DNE                     [4   bytes] (+12346  ms) [00 00 00 00] {ISO_STD_APDU}
000002 -- LOG_CODE_DNE                     [4   bytes] (+0      ms) [00 00 00 00] {ISO_STD_APDU}
000003 -- LOG_CODE_DNE                     [4   bytes] (+0      ms) [00 00 00 00] {ISO_STD_APDU}
000004 -- LOG_CODE_DNE                     [4   bytes] (+0      ms) [00 00 00 00] {ISO_STD_APDU}
000005 -- LOG_CODE_DNE                     [4   bytes] (+0      ms) [00 00 00 00] {ISO_STD_APDU}
000006 -- LOG_CODE_DNE                     [4   bytes] (+0      ms) [00 00 00 00] {ISO_STD_APDU}
000007 -- LOG_INFO_CONFIG_SET              [23  bytes] (~4969   ms) [49 53 4f 31 34 34 34 33 41 5f 53 4e 49 46 46 31 30 30 3a 4f 4b 0d 0a] {APDU: NONE RECOGNIZED}
000008 -- LOG_INFO_CODEC_RX_DATA           [1   bytes] (+8775   ms) [26] {DISABLE_VERIFICATION (ISO 7816-8), MFULC_REQA, REQA}
000009 -- LOG_INFO_CODEC_RX_DATA           [1   bytes] (+12     ms) [26] {DISABLE_VERIFICATION (ISO 7816-8), MFULC_REQA, REQA}
000010 -- LOG_INFO_CODEC_RX_DATA           [1   bytes] (+15134  ms) [26] {DISABLE_VERIFICATION (ISO 7816-8), MFULC_REQA, REQA}
000011 -- LOG_INFO_CODEC_RX_DATA           [1   bytes] (+16     ms) [00] {APDU: NONE RECOGNIZED}
000012 -- LOG_INFO_CODEC_RX_DATA           [9   bytes] (+0      ms) [93 70 87 d5 e1 f9 4a 09 e0] {MANAGE_CHANNEL (ISO 7816-4), INITIALIZE_PSAM (EN 1546-3), MFULC_SELECT_CL1, SEL_NVB_SAK_CRCA}
111111 -- LOG_INFO_CODEC_RX_DATA           [23  bytes] (~160    ms) [02 00 a4 04 00 0e 32 50 41 59 2e 53 59 53 2e 44 44 46 30 31 00 e0 42] {APDU: NONE RECOGNIZED}
222222 -- LOG_INFO_CODEC_RX_DATA           [16  bytes] (+237    ms) [03 00 a4 04 00 07 a0 00 00 00 04 10 10 00 9d 16] {APDU: NONE RECOGNIZED}
333333 -- LOG_INFO_CODEC_RX_DATA           [11  bytes] (+238    ms) [02 80 a8 00 00 02 83 00 00 18 2f] {APDU: NONE RECOGNIZED}
000016 -- LOG_INFO_CODEC_RX_DATA           [1   bytes] (+232    ms) [26] {DISABLE_VERIFICATION (ISO 7816-8), MFULC_REQA, REQA}
000017 -- LOG_INFO_CODEC_RX_DATA           [1   bytes] (+12     ms) [26] {DISABLE_VERIFICATION (ISO 7816-8), MFULC_REQA, REQA}
000018 -- LOG_INFO_CODEC_RX_DATA           [1   bytes] (+17     ms) [26] {DISABLE_VERIFICATION (ISO 7816-8), MFULC_REQA, REQA}
000019 -- LOG_INFO_CODEC_RX_DATA           [1   bytes] (+18     ms) [26] {DISABLE_VERIFICATION (ISO 7816-8), MFULC_REQA, REQA}
000020 -- LOG_INFO_CODEC_RX_DATA           [1   bytes] (+18     ms) [26] {DISABLE_VERIFICATION (ISO 7816-8), MFULC_REQA, REQA}
000021 -- LOG_INFO_CODEC_RX_DATA           [1   bytes] (+14104  ms) [26] {DISABLE_VERIFICATION (ISO 7816-8), MFULC_REQA, REQA}
000022 -- LOG_INFO_CODEC_RX_DATA           [1   bytes] (+12     ms) [26] {DISABLE_VERIFICATION (ISO 7816-8), MFULC_REQA, REQA}
000023 -- LOG_INFO_CODEC_RX_DATA           [1   bytes] (+6      ms) [26] {DISABLE_VERIFICATION (ISO 7816-8), MFULC_REQA, REQA}
000024 -- LOG_INFO_CODEC_RX_DATA           [1   bytes] (+6      ms) [26] {DISABLE_VERIFICATION (ISO 7816-8), MFULC_REQA, REQA}`

what it is? I did not send this (obviously) is it ATR ??
000012 -- LOG_INFO_CODEC_RX_DATA           [9   bytes] (+0      ms) [93 70 87 d5 e1 f9 4a 09 e0] {MANAGE_CHANNEL (ISO 7816-4), INITIALIZE_PSAM (EN 1546-3), MFULC_SELECT_CL1, SEL_NVB_SAK_CRCA}

my apdu looks different
I am sending 00 A4 04 00 07 A0 00 00 00 04 10 10 00 
but in the log I see 03 00 a4 04 00 07 a0 00 00 00 04 10 10 00 9d 16
where do the extra bytes come from, what do they mean?

I'm sniff log a communication card with a payment terminal makdonalds, there are many not signed apdu which cause interest, why they are not signed? because they are all from the emv standard as the rest
000050 -- LOG_INFO_CODEC_RX_DATA           [1   bytes] (~11     ms) [aa] {APDU: NONE RECOGNIZED}
000051 -- LOG_INFO_CODEC_RX_DATA           [1   bytes] (+0      ms) [4e] {APDU: NONE RECOGNIZED}
these bytes are too short to be an apdu command, there is no mention of them in the standards, where do they come from?

geo-rg, 2018-02-02:
Hi @hydra228 
I currently do not have the time to read your whole posting accurately, but the extra-bytes are ISO 14443 part 4, which encapsulates the APDUs. The first byte is sort of a header byte and in the end there is a CRC added. ISO 14443 part 4 also has some more bytes defined which may occur at some point after that first byte.

geo-rg, 2018-09-12:
@hydra228 I suppose your problem is solved. Please re-open this issue if not so.

closed 2018-09-12 by geo-rg