emsec/ChameleonMini #168: Another new GUI for RevG is coming up

gtpy, 2018-02-20:
Another GUI for ChameleonMini RevG is growing up...

Here the first screens:
[image]
[image]
[image]
[image]

Other features & GitRepo will follow soon

first impressions?

gtpy, 2018-02-21:
Repo is now available:
https://github.com/gtpy/ChameleonMini-RevG-GUI

iceman1001, 2018-02-24:
Nice one!   I like the idea with a picture of the device...   I'm gonna use that idea :)

maxieds, 2018-02-24:
Is this GUI going to be available on Linux at some point? It would make sense to use a portable GUI layer to implement the core functionality of the GUI and let the user choose where it gets compiled. I have tried to use Windows sanely, I really did, but made it 3 weeks before wiping the drive and giving myself a usable command line back. This would be **really** nice if it worked on my Debian-variant Linux box. :) Otherwise, for Linux only users, there's [my app](https://github.com/maxieds/ChameleonMiniLiveDebugger) on Android platforms.

iceman1001, 2018-02-24:
@maxieds   there is a GUI implemented in GO,    https://github.com/emsec/ChameleonMini/issues/167#issuecomment-367091177    that should work on linux.

geo-rg, 2018-03-22:
Thanks, @gtpy! I added a link in the wiki!

closed 2018-03-22 by geo-rg