emsec/ChameleonMini #170: Can it work with NFC-card programmer?

ellenhill94, 2018-03-16:
I am trying to use an NFC card programmer with the ChameleonMini to emulate a card. I have set up the ChameleonMini with the settings of the card I want to emulate (config = MF_CLASSIC_1K, UID = xxxxxxxx) but the card programmer cannot program it. I have also tried with UID=00000000. Other than the UID, the card is blank. 
Has anybody had success programming the ChameleonMini with a card programmer? Does a 'blank' card that can be programmed with the programmer actually have information on it that I am missing when I set up the ChameleonMini?
Thanks in advance

AndreasBujok, 2018-03-17:
Which NFC card programmer do you use?
I've done successful, i used a MFRC522 for example.
For debug you can switch the led to "field detect" mode. 

BR Andreas

geo-rg, 2018-03-22:
Hi @ellenhill94,
I suppose you mean writing data to the Chameleon-emulated card. This should be possible, but _maybe_ you have to upload a dump first, so the FRAM has a defined content (that you also know). You can find such dumps [here](https://github.com/emsec/ChameleonMini/tree/master/Dumps). Then I suggest using nfc-mfclassic as tool to write the card.

However, if you really mean _programming_ the Chameleon's firmware via the NFC interface, then no, this is not possible.

geo-rg, 2018-09-12:
@ellenhill94 I suppose your problem is solved. Please re-open this issue if not so.

closed 2018-09-12 by geo-rg