emsec/ChameleonMini #171: Why can't I change the setting number?

zhujiu, 2018-03-21:
Hi, Thanks for you open technology, I'm very appreciate for your generosity.
Nowdays ,I solder the pcb by myself. And yes it works with reading and emulating.
But  I found a problem, the partlist of device has a fatal mistake, 
BSS84AKW should change to    BSS84PW
BSS138BKW should change to  BSS138PW
and if you do  not change the part number, you can not read the uid number.

And here is my question,hahaha..
when I try to change the setting from 1 to other numbers,  it can't works,it still in 1 setting.
and there was  a  detail thing  I noticed，when I changed the settings, the COM Tera Term VT will disconnect form the chameleon and then connect it again.
So, I was stuck in here , please help me .And if anyone want to connect me in other detail things about chameleon  ,please send emial to me:  809687990@qq.com or Wechat number:  zhujiu00

zhujiu, 2018-03-21:
@doegox @robots @cons0l3 @0x715C @uhei

AndreasBujok, 2018-03-21:
I'm alitzle nur confused. Solder by yourself?
I know only about complete cards.

Cheers

zhujiu, 2018-03-22:
@AndreasBujok yes, i solder it by myself.
And does your chameleon could change the setting  for 1 to 2 ,  
like this command ;  setting =2
when i  set this command , and then check setting , the setting is still equal to 1. 
I do't know why?please help me .

zhujiu, 2018-03-22:
@AndreasBujok @doegox @robots @cons0l3 @0x715C

geo-rg, 2018-03-22:
Hi @zhujiu 
first of all: congrats for accomplishing to solder the Chameleon by yourself, we know that this is not very easy!
I suppose you use an ATxmega128a4u and that you are using a standard firmware here from the GitHub project. If not so, these are probable causes.
For changing the setting, it is necessary to access the eeprom, so you may also check whether you have any problems accessing the eeprom.

zhujiu, 2018-03-23:
@geo-rg actually  Im a hardware engineer， so  it's not so hard for me。 
yep， i'm not very famili with codeing。  and thanks for  your advice， i will chek  the eeprom 。

zhujiu, 2018-03-23:
@geo-rg  Could you tell the correct part number of the eeprom?
Does the device's name is MB85Rs256?  256KB flash?

zhujiu, 2018-03-29:
@AndreasBujok @geo-rg @doegox @robots @cons0l3

Peterthegreat, 2018-04-05:
> But I found a problem, the partlist of device has a fatal mistake,
> BSS84AKW should change to BSS84PW
> BSS138BKW should change to BSS138PW
> and if you do not change the part number, you can not read the uid number.
More information or reasons why? I am facing a similar issue but autocalibrate dose the trick.

zhujiu, 2018-04-18:
@Peterthegreat   contact me with my email 809687990@qq.com

Peterthegreat, 2018-06-27:
> @Peterthegreat contact me with my email 809687990@qq.com
@zhujiu No reply from that email address, but took the chance and gave it a try. Thank you so much! Now the read UID works. Will now test my other issues with the Chameleon

geo-rg, 2018-09-12:
I guess the problem is solved now. Please re-open this issue if not.

closed 2018-09-12 by geo-rg