emsec/ChameleonMini #174: Broken USB Connector (Please help fix it!)

maxieds, 2018-05-03:
Good morning. I have been testing out my [Android app](https://github.com/maxieds/ChameleonMiniLiveDebugger) and adding some nice new features to it. But before I can get them all working, the microusb connector broke off of the board (pretty flimsy to begin with: see the first image). Now it looks like in there are only three connections broken off and I noticed in the second image that there are three leads that are easily solderable. Can someone confirm / instruct me on which raw wires from a cut up microusb cable I can attach to these pads to get my Chameleon working again? Thank you very much!

[image]
[image]

Peterthegreat, 2018-05-03:
Hello @maxieds 
Didn't tried your app yet, but i'm looking forward to do that. Great work btw!

It looks like the pads are not peeled off. Put some solder on the pins then carefully clean it with solder wick. If the solder sticks to the pad, you're all good!

So, if the connector is intact, I would recommend to re-solder it. Use a lot of flux (e.g B-223) and high temperature 300 degrees on the cleaned soldering iron. Don't be afraid to take your time for each USB pin (about 5-7 seconds per pin) unless it feels like the pads are peeling off the PCB (in this case, reduce the temperature or solder the pin faster.

Good luck!

maxieds, 2018-05-03:
My soldering skills are very unpracticed and I'm working with an old iron from Radio Shack. Will it work to solder a pinned out USB cable to the big pads with lots of solder already on them? I know I can get that attached.

P.S. Since my Chameleon is broken until this weekend, I haven't been able to verify that the code improvements work with the board (just basic testing without USB on my dev phone). Would you mind testing out the new beta version [here (signed APK not yet on Play Store)](https://github.com/maxieds/ChameleonMiniLiveDebugger/releases/tag/0.3.7-free), especially the DUMP_MFU printing and cloning features?

ceres-c, 2018-05-03:
You might need some flux, just in case you end up connecting 2 pins. If you're going to wire the usb to pads it shouldn't be needed, as it's easy to do them properly at first try

Peterthegreat, 2018-05-04:
> Will it work to solder a pinned out USB cable to the big pads with lots of solder already on them

If you want to go with that option, it's very recommended to glue the usb cable to the PCB with a glue gun. This way, you won't stress the PCB pads and end up breaking them. 

> Would you mind testing out the new beta version here (signed APK not yet on Play Store), especially the DUMP_MFU printing and cloning features?

If I get the old firmware working, I would gladly test it. Hope I can find some time this weekend.

maxieds, 2018-05-07:
I'm still trying to get my hands on a working soldering iron. Hopefully I will be able to get one tomorrow. Please see [this issue](https://github.com/maxieds/ChameleonMiniLiveDebugger/issues/10) in the meantime. If anyone would be kind enough to check that the latest beta release of my app is working correctly, then I can release it to production. I'd really appreciate the help testing!