emsec/ChameleonMini #177: compatible with rev E Reboot ? or not ?

Shashadow, 2018-06-07:
Hello,

just a little question about firmware.
you speak about Rev E and Rev E light, but not about Rev E Rebooted.
I have a Chameleon Rev E Rebooted, and because firmware of this one doesn't support mifare classic with 7bit UID, I would like to know (before to make a mistake :-) ) if your firmware for revG work for revE ? do you know if it's compatible ?
or if I must wait for a emulation 7bit with official release for rev E Rebooted

Thanks

ceres-c, 2018-06-07:
RevE Rebooted was not made by original Chameleon devs, so they're not officially supporting it. It uses another SOC (even if it's speculated to have higher memory, as of https://github.com/iceman1001/ChameleonMini-rebooted/issues/42) and some funny flashing procedure (https://github.com/iceman1001/ChameleonMini-rebooted/issues/11)

Shashadow, 2018-06-07:
Very interesting for the first link.
Ok thank you for information, I will waiting for new firmware from iceman.

closed 2018-06-07 by Shashadow