emsec/ChameleonMini #178: problems with reading on acr122u

gtpy, 2018-06-08:
If I put a for example a classic card on acr122u the reader can identify it.
If I configure the cham (RevG) to MF_CLASSIC_4K_7B or other the acr122u couldn't identify it correct.
No stable connection is possible. With other tags no problem only with the cham.

Any ideas how to solve it?

gtpy, 2018-06-20:
I tested the reaction times with a reader at a printer and there it is really slow.

Already used autocalibrate.
At the beginning with an older firmware version it worked better.
At the moment I couldn’t find the problem of it.

Any ideas how to get it better?

Peterthegreat, 2018-06-28:
Autocalibrate is part of the reading process, it sets the correct threshold according to the current card. The emulation is not affected.
- try another reader 
- try  battery mode 
- find the older, working fw

geo-rg, 2018-09-12:
@gtpy Can you tell me which older firmware did work better?

gtpy, 2018-09-12:
@geo-rg  I'll try to test it the next days or at the weekend.
You'll hear from me.

gtpy, 2018-12-10:
Lost sight of this issue. I will soon deal with it again.