emsec/ChameleonMini #184: Mifare Classic Clone + Reader Attack

X0x1RG9f, 2018-09-18:
Good Afternoon,

I just bought a Proxmark3 RDV4 and the Chameleon Mini RevE Rebooted. I'm starting to play with both, but  I'm a little stopped in my enthusiasm with very simple things on Chameleon. I don't know if it could be a chameleon hardware issue or simply one thing I did not understood, so I will try to explain my best with what I know !

### 1st Issue : Cloning a Tag
I have a simple tag for entering my building which is a MIFARE CLASSIC 1K, as show below :
```
proxmark3> hf search
hf search

 UID : be ** ** **
ATQA : 00 04
 SAK : 08 [2]
TYPE : NXP MIFARE CLASSIC 1k | Plus 2k SL1
proprietary non iso14443-4 card found, RATS not supported
Chinese magic backdoor commands (GEN 1a) detected
Prng detection: WEAK

Valid ISO14443A Tag Found - Quiting Search
```
Copying that tag on a simple Magic Chinese Card works fine with the bellow commands, and I can enter the building : 

```
hf mf mifare
hf mf nested 1 0 A ************ d   (*** being previous found key)
hf mf dump
hf mf restore 1 (on a BLANK card)
hf mf csetblk 0 xxx...xxx (I set it manually because for unknown reason it cannot be written with restore)
```
This is working fine, and I would like to reproduce exactly the same behavior with the Chameleon. However, after some days trying and trying, whatever GUI I use, its ends in a big failure :(.

My steps in the GUI are the followings : 
1. Connect to the Chameleon (Firmware Chameleon-new-1.0) in the GUI
2. In "Operation" tab, select "Slot 1", and then select MF_CLASSIC_1K
3. I then click on "Upload Dump" and upload the previously created dumpdata.bin with proxmark
4. UID is then replaced by my dump
5. I click on "Apply" 

At the contrary of Magic Chinese Card, this is not working at all with the building reader(s). When I approach the Chameleon from the reader, red light of the TAG1 is ON (sometimes fixed, sometimes blinking, can't explain). Readers, on their sides, sometimes pop a red light, or most often nothing, like if nothing was read from Chameleon. 

Also, when I try to read the Chameleon back with the Proxmark, I sometimes have some reading errors : 

```
proxmark3> hf mf dump
hf mf dump
|-----------------------------------------|
|------ Reading sector access bits...-----|
|-----------------------------------------|
[...]
#db# READ BLOCK FINISHED
#db# READ BLOCK FINISHED
#db# Cmd Error: card timeout. len: 0
#db# Read block error
#db# READ BLOCK FINISHED
#db# Can't select card
#db# READ BLOCK FINISHED
#db# READ BLOCK FINISHED
|-----------------------------------------|
|----- Dumping all blocks to file... -----|
|-----------------------------------------|
[...]
Successfully read block  2 of sector  7.
#db# READ BLOCK FINISHED
Successfully read block  3 of sector  7.
#db# READ BLOCK FINISHED
Successfully read block  1 of sector  8.
#db# Can't select card
#db# READ BLOCK FINISHED
#db# READ BLOCK FINISHED
Successfully read block  2 of sector  8.
#db# READ BLOCK FINISHED
[...]
Successfully read block  0 of sector 10.
#db# Authentication failed. Card timeout.
#db# Auth error
#db# READ BLOCK FINISHED
#db# READ BLOCK FINISHED
Successfully read block  1 of sector 10.
```

Am I missing something here ? I must admin that I have no idea of what can I be doing wrong, any help could be appreciated on that subject. 

### 2nd Issue : Reader Attack
I also tried the reader attack presented in that video (https://www.youtube.com/watch?v=1VpXC3-eKhc), configuring the "Slot 2" with MF_DETECTION and inserting a valid card UID.

When I approach from the readers again, same behaviour, sometime fixed red light, sometimes blinking.

I then come back on the GUI and tries a mfkey32 on the "Slot 2", but again, nothings happens and absolutely nothings appear on the screen. 

Any idea on that behaviors also ? 

I'm a beginner on these subjects, and your help would be really appreciated. I don't have any problems with the Proxmark, but actually I cannot do anything with the Chameleon ! 

Thanks for your attention and your help !

gtpy, 2018-09-18:
@X0x1RG9f 
this official repo is for RevG board.
Your device (RevE Rebooted) is a modified one and not the original.
Your Question is right at the other repo: [Chameleon-Mini RevE Rebooted](https://github.com/iceman1001/ChameleonMini-rebooted/issues/57)
I think the community there is more in touch with this board.
Wait some days - I think you'll get there a solution.
These guys are really great at help and developing for RevE Rebooted (firmware and GUI)

If you get a RevG board - we can help you here :-)

X0x1RG9f, 2018-09-18:
Sorry for posting here then, I will wait an answer from the other repo ! Thanks for your fast reply. Actually the website where I bought it is redirecting to this repository, maybe a mistake !

gtpy, 2018-09-18:
No problem I’ll try to take a look at your question in the other repo.
Didn’t own a RevE rebooted device so I could only try to help theoretically.

Pls close this issue here.

closed 2018-09-18 by X0x1RG9f

yaxax, 2019-03-02:
I can not get out with the proxmark3.I try to make a master key but I can not. I also try to get info from the reader but I can not. I am new in this I am learning can someone helping me with that and explained me how thanks.