emsec/ChameleonMini #185: Sniffer not workin in metro!

hydra228, 2018-09-23:
I used the sniffer to read the paypass there everything is visible, but I clicked into the underground metro to check how the turnstile communicates and my card, the log is empty, I do not see a single command from the card to the turnstile: https://hastebin.com/ewumapayix.cs

why do I see the once command - [93 70 11 55 55 55 44 e8 ce]?
11555555 my card uid yes? but at least the turnstile is required to record its serial number on the card, update the number of trips on the card, and if possible, then reduce the balance.

gypsophlia, 2018-09-24:
I think you need to do auto calibrate in sniffing mode.  It will set the threshold which is suitable for the card->reader trace capture.  The autocalibration in sniffing mode monitor the anti-collision process, so please also make sure that there are several anticollision processes when the card is read. 

If there are not any, you can try adjusting the threshold manually so that the card->reader traces can be captured. 

I remember there is an android app which can be used for controlling ChameleonMini in the filed, but you may need to modify that to make it support sniffing/ setting threshold during sniffing.

geo-rg, 2018-09-25:
Hi @hydra228 
I noticed that you use the app developed by @maxieds, which is great. However, I suppose that his app does not yet support the new log codes for sniffing both directions. Then I could imagine that the app drops log messages if it does not know the type code (please correct me if I'm wrong, @maxieds). In this case, sniffing would have been successful but you cannot get the information. So please try the chamlog tool once to check this.

Additionally to what @gypsophlia has suggested you may also try out different positions of the Chameleon to get a better sniffing result.

maxieds, 2018-09-27:
Thanks to one of my users, this unread thread from a long list of waiting emails in my inbox was brought to my attention. I am currently working on adding the new logging codes to the source repo for my app. I should have something ready to push to the Play Store site for users by later tonight / sometime this morning. 

@hydra228 
If I can get a few users to test and verify that my implementation of the new logging codes and features works for you, I would appreciate the input and feedback. 

I will post here again when I have pushed the new code for users to install.

maxieds, 2018-09-27:
The new logging codes implemented in the firmware [here](https://github.com/emsec/ChameleonMini/blob/master/Firmware/Chameleon-Mini/Log.h#L26) have been added into the log parsing routines in the [latest release (v0.5.4)](https://github.com/maxieds/ChameleonMiniLiveDebugger/releases/tag/v0.5.4) of the Chameleon Mini Live Debugger app. I have pushed the new APK files to the Play Store. These should be available to users over the next few hours. 

Thanks again to Cesar, who always appends **[RUSSIA]** to his emails, for pointing this out to me. Can @hydra228 and Cesar, to have at least two testers, verify that this problem has been fixed now? If you all can post a verification here below, I will check this thread to resolve any remaining issues. I'm glad that you all enjoy my app. Please drop me a line if there are still any missing features you would like to see, or just to let me know what new uses you have for the app. It will be much appreciated and keep me motivated on posting improvements in the future. :green_heart: :smiling_imp: :exclamation:

maxieds, 2018-09-27:
To those of you who are testing out the new features to resolve this bug in the app, please do note that you will need the latest app (v0.5.4) **AND** the latest ChameleonMini firmware loaded on your Chameleon board for this to work. This slipped my mind for a little bit, but should probably be repeated anyway.

hydra228, 2018-09-28:
@maxieds, hydra228 == Cesar, it's me :)) all one)
Friends I'm going to go and collect a lot of logs, I want to update the firmware, tell me I'm taking the firmware from the ChameleonMini / Firmware / Chameleon-Mini / Latest folder / is this the right solution?
The version command produces the following:
101:OK WITH TEXT
ChameleonMini RevG 180912 using LUFA 151115 compiled with AVR-GCC 5.4.0. Based on the open-source NFC tool ChameleonMini. https://github.com/emsec/ChameleonMini commit 10144f1

maxieds, 2018-10-08:
@hydra228 
I believe you confirmed over email that the app now correctly sniffs the data you were looking for. Is that correct? Is there still any data you cannot retrieve with the app?

hydra228, 2018-10-08:
@maxieds 
Hi colleague!!
it's hard for me to say for sure.
I still could not find a card key in logs, but I think they should be there.

This weekend I will go to the subway again, I will emulate two identical situations with a card, one case will use your program, the second case will use instructions for a laptop, I will send both dialogs to you!

monster1025, 2020-06-25:
Still not figured it out?

Here is my logs from Russian subway (metro):
https://pastebin.com/tbgYRPP2

Dump and auth keys are correct, Chinese white card with same data are accepted.