emsec/ChameleonMini #186: Sniffer

nurzhannn, 2018-09-24:
Hello
how to remove the restriction  SNIFFER CARD -->READER  and intercept all data with command CONFIG=ISO14443a_SNIFF ?

gypsophlia, 2018-09-25:
Please upgrade to the latest firmware

nurzhannn, 2018-09-25:
> Please upgrade to the latest firmware

i have "reve-2" it stands there atxmega32a4u,will the new firmware work?

geo-rg, 2018-09-25:
@nurzhannn Sorry but no, the sniffing firmware is only for RevG.

nurzhannn, 2018-09-25:
this restriction is specially done?

nurzhannn, 2018-09-25:
i bought for the sake of the sniffing,what should i do with him ?

nurzhannn, 2018-09-25:
American capitalists :)

ceres-c, 2018-09-25:
The code is here available, you are free to backport it into the other repo. Easy, innit? ;-)
Or you can buy the revg which is already supported by new code and will be supported in the future...

nurzhannn, 2018-09-25:
after firmware Revg need to calibrate, yes ? :-)

geo-rg, 2019-03-01:
Hi @nurzhannn 
if you do not have a RevG Chameleon by Kasper & Oswald, I cannot help you here. Earlier in this thread you stated that you have a RevE (rebooted?) chameleon.

If you have a RevG by Kasper & Oswald, you may re-open this issue. Else, please contact the manufacturer of your Chameleon.

closed 2019-03-01 by geo-rg