emsec/ChameleonMini #187: Feature request: extending / modifying the SEND and SEND_RAW commands

maxieds, 2018-09-27:
Hello. My Android app, the [Chameleon Mini Live Debugger](https://github.com/maxieds/ChameleonMiniLiveDebugger), has some features to send and recognize arbitrary APDU commands. In particular, there is a whole tab in the application dedicated to this (see the attached screenshot). One extension of the *SEND* and *SEND_RAW* commands currently in the Rev. G board firmware which would be **extremely useful** to me and including some nice new features to go along with the app, would be for these commands to accept / parse hexadecimal byte strings of **arbitrary** (or say reasonably bounded) length instead of just a single XX byte input. 

[image]

For example, there are some older commandline (and not specifically Chameleon Mini oriented) utilities which can do discovery on NFC tags, door locks, etc, by sending the device a sequence of standard APDU commands (multiple bytes a piece for reference) and parsing the expected responses. I would like to add in some new discovery abilities with the Chameleon Mini in my app along these lines. For example, we could potentially add in some scripting prompted by user configuration which will allow an APDU command to be send with the modified *SEND*/*SEND_RAW* commands I have requested, and then give a list of rules for what to send back to the device given it's response to the previous commands. Like I said there is some definite new possibilities for doing discovery and live hacking (pen testing if you prefer) with the Chameleon if only we had commands in the firmware to send variable length APDU commands from the board. 

Is this a feasible request? How difficult would this be to add into the device firmware?

geo-rg, 2018-09-27:
@maxieds If I understand correctly you mean something like "send 9320" to work? This is already implemented. Both `send` and `send_raw` accept up so 256 (not so sure whether this is the exact value) bytes. Have a look at the examples in https://rawgit.com/emsec/ChameleonMini/master/Doc/Doxygen/html/_page_14443_a_reader.html.

maxieds, 2018-09-27:
Following from the DESFire (EV1) examples listed [here](), I would need something more like *SEND afb0ccbced8fc838c908dce24d86caec3c* to work. That should be a good upper bound on the number of bytes (say, +7 to this count) that I would need for the *SEND* / *SEND_RAW* commands to support. Is this already built into the firmware. If so, I will go ahead and start tinkering with the APDU-based detection code on my end.