emsec/ChameleonMini #188: Questions on memory types (RevG)

hsanjuan, 2018-09-30:
Hello, I have some questions on "memory" (RevG). Docs are not super helpful and I wonder if my intuition is correct. If there is a place where this is explained, please point me to it...

There is an SRAM/Non-permanent memory, sometimes referred as `Flash` and there is an FRAM/Permanent memory, which is also a Flash anyway. Correct? Are there more memories involved when operating the RevG?

Are these both "external"? Are the internal memories of the chip used for something (other than holding the firmware)?

How big is the FRAM. I read ` 256 kBit` in kickstarter but `logmem?` reports `18430 (from which 16382 non-volatile)`. Is the log space limited and the rest is used to store cards?

And how big is the SRAM? Can it hold 4K mifare cards? (logmem suggests it's ~2K only)

`RECALL` the default lbutton action (store ` the content of the current slot from Flash to FRAM`) but I don't understand on which occassion it should be used. [This](https://rawgit.com/emsec/ChameleonMini/master/Doc/Doxygen/html/_page__settings.html) seems to indicate that settings are loaded and saved from/to FRAM automatically (on change at least). Do `STORE` and `RECALL` need to be called at all then? For what usecases?

It seems sniff logs are written to SRAM but it is small (2K?). Does `LOGSTORE` store sequentially in FRAM, meaning it can be triggered several times and it will sequentially fill FRAM up? Does it need the manual `logstore` or will it automatically dump itself to fram when it gets full?

Thanks

gtpy, 2018-10-01:
@geo-rg maybe you can help here?

geo-rg, 2018-10-01:
Hi @hsanjuan 
let me explain the memory architecture of the RevG ChameleonMini. It uses 4 different memory types:
internal:
1. Flash (128 kB non-volatile)
2. EEPROM (1 kB non-volatile)
3. SRAM (8kB volatile)
external:
4. FRAM (256 kBit = 32 kB non-volatile)

**Flash**
The lower half of the flash is used for the firmware, the upper half (=64 kB) is used to store card contents permanently. As you might have noticed, there are 8 slots/settings on the Chameleon. Thus, every setting can store up to 64 kB / 8 = **8 kB**. However, since reading from/writing to flash is slow, this is only used for permanent storage.

**FRAM**
For emulation, the very fast FRAM is used. This has a size of 32 kB from which the first 8 kB are used for emulation. On every setting change, the data from FRAM (first 8 kB) are stored to the flash memory and the data from the new setting are read from flash to FRAM.
The subsequent memory area of 8 kB is currently unused and finally the upper half of FRAM (=16 kB) is used for logging.

**SRAM**
This is the fastest memory (minus GPIORs and internal registers), here are all the intermediate variables and buffers stored.

**Logging**
The log functionality works like this: When a log entry shall be written, this is done in an area of 2048 bytes within the SRAM. Every 100 ms the Chameleon checks whether there are new log entries in SRAM and if so, it stores them to the FRAM log area. If the FRAM log area is full, the SRAM log area is filled as well. When typing `logmem?` you obtain how many log memory is free in total (SRAM + FRAM log area) and how many non-volatile is free (FRAM log area).

**store, recall and logstore**
`logstore` was necessary "historically" when the Chameleon did not check itself every 100 ms whether to store the log from SRAM to FRAM.
If you emulate a card, let's say MF Ultralight, and the reader does a write, the contents on FRAM is changed. When you power-cycle the Chameleon without changing the setting and without running `store`, the FRAM content is overwritten on power-up by the flash memory (which has not been altered). So if you want the reader's write command to be permanent to the emulated card, you would have to run `store` or change the setting.
`recall` is the exact opposite. If you want to reverse the impact of the reader's write command on the emulated card's content, you can run `recall` (or power-cycle the Chameleon).

I hope this answers all your questions :)

hsanjuan, 2018-10-01:
Thank you @geo-rg , this is simply GREAT!

AndreasBujok, 2018-11-16:
@geo-rg 
Great explanation, these are very interesting explanation.
I will add these next on my "Userfriendly Docu" :-)

Thanks

closed 2019-02-28 by geo-rg

eXspir3, 2019-04-10:
This clarifies things quite a lot thank you :)!

I have a question to add here.
To my understanding when i use ISO_14443_SNIFF Mode, set logmode to memory and place the Chameleon between reader and tag it should sniff the traffic and then save it to SRAM and then FRAM.

But how can i retrieve what was sniffed?

Best Regards :)

geo-rg, 2019-04-10:
Hi @eXspir3 
you can use `logdownload` to download the log via xmodem (further details depend on your terminal program). You can decode this program with `python3 chamlog.py -f path/to/logfile`.