emsec/ChameleonMini #190: How to read the logs

sp4rrows, 2018-11-15:
Hello everybody,
I'm new to nfc.
Does anyone have a reading to learn general and how logs are read interpreted?
I'm playing around with my new chameleon but need some point to start.

Thanks a lot.

geo-rg, 2018-11-16:
The log can be downloaded from the Chameleon either with a terminal emulator and the Chameleon command `logdownload` or with the [chamlog tool](https://github.com/emsec/ChameleonMini/tree/master/Software).

sp4rrows, 2018-11-16:
@geo-rg  thanks that worked. 
Is there any hint, book, documentation how I can learn more about these logs in general.
I’m not so familar in rfid. Need to learn it on my own.(Also I am a beginner in programming)
Example: 
How to read them / understand them?
Thank you.

maxieds, 2018-11-16:
If you are looking for a user friendly GUI on an Android phone, you can
check out my app: https://github.com/maxieds/ChameleonMiniLiveDebugger

sp4rrows, 2018-11-16:
Thanks. I’ll go on search and try it out.

closed 2018-11-16 by geo-rg

gtpy, 2018-11-16:
For understanding the chameleon it is good to read the 
https://rawgit.com/emsec/ChameleonMini/master/Doc/Doxygen/html/_page__getting_started.html
Then try out and use command line to understand it better.
Then you could use the GUI and Android tool which is a great help.

AndreasBujok, 2018-11-16:
That's also my stand on the beginning - i was total confused. I'm not familiar with the kind of doxygen docu.

After growing of the "HowTo" i've written an own docu: https://github.com/AndreasBujok/ChameleonMini

Could be helpful for your request :-)

BR Andreas