emsec/ChameleonMini #193: Config of buttons individual for each slot

Vrumfondel, 2018-11-30:
Wouldn't it make sense to have individual button-settings for each slot? Right now they seem to be global.

geo-rg, 2018-11-30:
Hi @Vrumfondel,
whether this makes sense or not depends on your use-case. For example, if you want to use the buttons to switch through slots, a different button-setting for each slot could lead to the situation that you enter this slot by pressing a button, but you cannot leave it anymore (without terminal) since this slot has another button setting.

Anyways, this is already possible, you can simply change [these lines](https://github.com/emsec/ChameleonMini/blob/master/Firmware/Chameleon-Mini/Makefile#L42-L43) and recompile.

closed 2018-11-30 by geo-rg

Vrumfondel, 2018-11-30:
I see, and agree that use of global or individual settings depends on use-case.
Then, what of making that configurable instead of only compile option?

Vrumfondel, 2018-12-03:
Has this feature using the compiler-switches really been tested? To me it seems not to work - the settings are always global ......

Vrumfondel, 2018-12-04:
PR done .....

herrmanns, 2019-01-27:
working! many thx vrumfondel!

herrmanns, 2019-01-28:
another idea.

how about completely individual_configurable UID. maybe somebody of you guys have such solution already.
my idea was to add the following (hopefully understanable)

[11] 22 33 44 55 66 77
Lbutton 
[12] 22 33 44 55 66 77 (that's of course possible already)
Lbutton 
[13] 22 33 44 55 66 77
Rbutton  (here starts the idea)
Lbutton
13 [23] 33 44 55 66 77
Lbutton 
11 [24] 33 44 55 66 77
Rbutton 
Lbutton
13 24 [34] 44 55 66 77
Lbutton
13 24 [35] 44 55 66 77
Lbutton_long
Lbutton
13 [25] 35 44 55 66 77

Lbutton == one up
Rbutton == one right
Lbutton_long == one left

Vrumfondel, 2019-02-11:
Well,
I think I get your idea, but I see a big problem in the usability: I don't really see a good way to feed back which byte of the UID is currently being changed - not even talking about the actual set UID.

The pure implementation is not a real issue ...

herrmanns, 2019-02-11:
hi vrumfondel,

yes, it is a bit tricky to show whats going on with just 2 LEDs. it may be possible to use the LEDs and "long button press" for it.
and you have to store the current hex in your brainware :) or you are able to connect some raspi with display to read it out (if really necessary).
in my use case i dont need to know that. the only thing which is important is that i know which "hex slot" is currently active to change.