emsec/ChameleonMini #197: Bricked after failed firmware upgrade

jwgn, 2018-12-14:
Hi,

I tried to perform a firmware upgrade, but it failed:

Creating the EEPROM binary...
Write done!

Creating the Flash binary...
B:\Downloads\avr-objcopy.exe:B:\Downloads\Chameleon-Mini.hex: File format not recognized
Not find fileB:\Downloads\flash.bin konnte nicht gefunden werden

Flashing the files onto the "Chameleon-Mini Rev-E Rebooted"...
old_driver_bootloader
Erasing flash...  Success
Checking memory from 0x0 to 0x6FFF...  Empty.
0%                            100%  Programming 0x80 bytes...
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
0%                            100%  Reading 0x400 bytes...
not_find_file

If there are no errors above, flashing the firmware to your "Chameleon-Mini Rev-E Rebooted" should be finished now. Enjoy!

Now I am unable to restore the Chameleon. It starts directly into bootloader mode when I connect it via USB (no LED flashing, shows up in device manager as ATxmega32A4U) but it is no longer being recognized by the flashing tools:

B:\Downloads>BOOT_LOADER_EXE.exe
old_driver_bootloader
dfu-old-driver: no device present.

Any Idea on how to get it back to work?

ceres-c, 2018-12-14:
Looks like you own a ChameleonMini Rev E (the cinese version, by the way). This repo is not the right place where to ask, you have to go over at iceman1001/ChameleonMini-rebooted and ask there.
By the way, have you tried putting the Chameleon in upgrade mode (right button pressed while connecting the USB, if I'm not mistaken) and trying to flash again?

jwgn, 2018-12-14:
Oh sorry, you are right, this is not the correct place to ask. I will head over to the correct Github page.

To answer your question: Yes, I tried that, but it automatically boots into bootloader mode anyway. Does not make any difference when I hold the button.

closed 2018-12-14 by jwgn