emsec/ChameleonMini #198: Not Able to Transmit

herrmanns, 2019-01-08:
Hi, I have the Cameleon mini revG nad it works like charm except the writing/clone mode. It seems everything which has to do with transmitting from device to some reader doesnt work.
when i hold it on to a reader nohting happens. i tried to flash firmware which didnt make a difference.
I got the revG 180912. any help appreciated.

gtpy, 2019-01-08:
Can you give a bit more input.
Which type of config you are using?
Type of Reader?

herrmanns, 2019-01-08:
i tried to read with proxmark3rdv2 also with rdv4. currently i tried it with an acs acr122 reader.
not one card emulation mode works with chameleon.
lets say i set mode to MF_CLASSIC_1K and UID BB9944AA and hold it to any of the reader it just act like nothings there. really odd. i had the chameleon in a box for some months and it was working like charm. but now i got this problem. i have no clue what happened to it.

geo-rg, 2019-01-09:
Hi @herrmanns,
did you buy your Chameleon from Kasper & Oswald or a remake from another vendor? Please note, that some of the remakes do not work well and we cannot give support for this since we do not know any changes they made in the internals.

However, if you bought your ChameleonMini from Kasper & Oswald: Can you please set `logmode=memory`, then `config=MF_CLASSIC_1k`, then hold the Chameleon to the acr122 and run `nfc-list` from libnfc. Then download the log via `logdownload` or with the chamlog tool and finally decode the log with the chamlog tool (if you need any help for these last steps, I can also give you more detailed instructions). Then, post the log here, so we can see what goes on.

herrmanns, 2019-01-09:
yes it is the original from kasper n oswald. 
hm thats exactly the problem. no reader could read anything because chameleon is not transmitting at all (or the signal is so weak for some reason that no reader could react).
the antenna and also the antenna connection seems to be fine. because when i turn the chameleon to act like a reader with mode ISO14443A_READER then there is no problem at all. i could hold the card 3 - 5 cm away and chameleon is reading with out any trouble. is there any other firmware i could try (newer)?
--> version?
101:OK WITH TEXT
ChameleonMini RevG 180912 using LUFA 151115 compiled with AVR-GCC 5.4.0. Based on the open-source NFC tool ChameleonMini. https://github.com/emsec/ChameleonMini commit 10144f1

ceres-c, 2019-01-09:
Hi,
I had a similarly curious problem some time ago and it was a PEBKAC error. I had placed the reader on my mouse mat and, for the life of me, couldn't get the chameleon to respond in any way. As it turns out, my mouse mat is made of aluminium, but I couldn't figure it out until I put my hand on it.

I am not saying this is your case, but have you tried to read the chameleon with you cell phone moving away from your current setup?

geo-rg, 2019-01-09:
Hi @herrmanns,
your firmware is up to date. Please set `ledred=codec_rx` and re-try. If the red LED blinks when you hold the Chameleon to the reader, it definitely gets some request (there is no information whether this request is understood correctly, however). As a next step, you could set `ledred=codec_tx`, which would blink, if the Chameleon even responds. In any way it would be useful if you could enable the log and post the decoded log here.

herrmanns, 2019-01-09:
thats the codec_tx log:
https://pastebin.com/VfbWTtCu

thats the codec_rx log:
https://pastebin.com/RF8sfdpv

btw thank you so much for helping me. i m really lost.

geo-rg, 2019-01-09:
Hi @herrmanns,
in both cases, the Chameleon receives and decodes correctly the reader's request and also sends an answer that the reader assumedly does not understand. Please try different setups. Maybe it is really a problem like @ceres-c explained. Have you tried reading the Chameleon with a smartphone?

herrmanns, 2019-01-09:
thats codec_tx first row and second row is rx again but this time i used pm3rdv4 as a reader. hf 14a read.
https://pastebin.com/x2MuzEsy
https://pastebin.com/76VQ4tDu

i was also on the "public real live test reader" and nothing detected. it is just quiet. i coul hold dirty toilet paper on it and i would get the same result (almost) :) .

it is terrible i like this device!

geo-rg, 2019-01-28:
Hi @herrmanns,
since you posted on #196, I assume you solved the problem? If not so, please contact chameleon@kasper-oswald.de and we'll find a way to repair or replace your Chameleon.

closed 2019-01-28 by geo-rg

herrmanns, 2019-01-30:
yep, problem solved. it was the hardware. a dry solder joint. an extremly rare issue for a chameleon which is a fantastic product... but in general not completely impossible. 
my other units running smoothly. the chameleon guys fixed it. thx for that btw !  :D

magicbugpatrol, 2019-11-06:
I have a similar problem. Emulation didn't work.
Reader work.
Tested:
ledred=codec_rx
ledred=codec_tx 
How to test further here?
Thanks.

fptrs, 2019-11-08:
Hi @magicbugpatrol,
can you describe your setup (the reader you use, configuration on your chameleon, ...) and provide the chameleon log?

magicbugpatrol, 2019-11-09:
Hi @fptrs,

configuration:
`config=MF_CLASSIC_4K_7B`
`100:OK`
`uid=01020304050607`
`100:OK`
`uid?`
`101:OK WITH TEXT`
`01020304050607`

I use my phone as a reader.
My phone read with NXP taginfo my original classic cards but not the chameleon.
If I configure the chameleon setting to reader it can read the UID of the original card.

With above configuration and 
with `ledred=codec_rx` and `ledred=codec_tx`setting the red LED flashes.

Set logmode to memory and get the chameleon log with chamlog.py
https://pastebin.com/6en34Nvj

fptrs, 2019-11-11:
Hi @magicbugpatrol,
in the log you can see that the Chameleon answers with the ATQA to a REQA, so far so good. But afterwards it does not receive a valid command (e.g. SELECT_CL1). Can you try to use an additional reader and provide the corresponding log as well?

magicbugpatrol, 2019-11-11:
Hi @fptrs,
Thanks for the reply.
I was able to borrow an ACR reader from a colleague.
https://pastebin.com/RR3nRFAE
Don't get better results.
Are there other ways to test the functionality?

fptrs, 2019-11-11:
Hi @magicbugpatrol,
sounds like a hardware problem. Where did you order your Chameleon?

magicbugpatrol, 2019-11-11:
Hi @fptrs,
I bought it at Kasper & Oswald (their web shop)
It's been over a year, though.

fptrs, 2019-11-18:
Hi @magicbugpatrol,
since a hardware problem is very unlikely, can tell me which firmware you are using and what the "version?" command answers?

magicbugpatrol, 2019-11-19:
Hi @fptrs, 
Now I use the latest firmware. 
No change to previous version.
`version?
101:OK WITH TEXT
ChameleonMini RevG 191119 using LUFA 151115 compiled with AVR-GCC 5.4.0. Based on the open-source NFC tool Chameleon
Mini. https://github.com/emsec/ChameleonMini commit a35610f
`

Same problem.

Configuration as reader works:

`setting=2`
`100:OK`
`config=ISO14443A_READER`
`100:OK`
`getuid`
`101:OK WITH TEXT`
`00DDDD1B`
`identify`
`101:OK WITH TEXT`
`MIFARE Classic 1k`
`ATQA:   0400`
`UID:    00DDDD1B`
`SAK:    08`

magicbugpatrol, 2019-12-03:
Thanks. Was fixed by Kasper & Oswald.