emsec/ChameleonMini #200: Issue not sure how to resolve?

OlivierLaflamme, 2019-01-11:
when I Run the ChameleonFirmwareUpgrade.bat as Administrator I get this error message in my terminal?

dfu-programmer: no device present.
There was an error with executing this command. Maybe your ChameleonMini is not in bootloader mode?

Do you happen to know what I could do to fix this?

OlivierLaflamme, 2019-01-11:
how would  I put it in bootloader mode?

gtpy, 2019-01-11:
> how would I put it in bootloader mode?

Make sure the Chameleon is switched off (power switch on the side is in the position OFF). Now, hold down the right button (RBUTTON) and keep it pressed while connecting the Chameleon to a PC via a USB cable.

Best way to start is to read the [getting started](https://rawgit.com/emsec/ChameleonMini/master/Doc/Doxygen/html/_page__getting_started.html)

geo-rg, 2019-01-11:
@OlivierLaflamme Does this solve your problem?

OlivierLaflamme, 2019-01-11:
yeah thanks soz for question

closed 2019-01-11 by geo-rg