emsec/ChameleonMini #202: Problem setting config to vicinity

geo-rg, 2019-01-16:
Hi,
I have a problem with the chameleon mini rev G. 
[image].

What can i do to fix this?
I tried to upload new firmware 3 times.
May someone has an idea.
Thanks

_Originally posted by @jan1702 in https://github.com/emsec/ChameleonMini/issues/199#issuecomment-454772263_

geo-rg, 2019-01-16:
@jan1702 In order to check which configurations are available, you can type `config=?`. As you will find out, VICINITY is not an option, since this is not yet supported by the firmware. However, this will be soon available and if you want to try it out already, you can have a look here: #182 

You can check out this pull request and compile this firmware by yourself to try the vicinity config.

closed 2019-01-16 by geo-rg

ceres-c, 2019-01-16:
You can use my repo, but notice Vicinity hasn't been actually implemented, it's mostly a stub
The easiest tag you can find is Titagitstandard, while EM4233 has more complete and better engineered code