emsec/ChameleonMini #203: Reader attack + GUI tools + Sniffing

ictmehrdad, 2019-01-29:
Hello
I have a ChameleonMini REV-G and Have 2 problem :
1- After I Clone the Project;How Can Run that to see the GUI?
2- How Can I read the Reader then From it Find the Key and Clone a card To open that?(For Both Mifire 1K and 4K ; encrypted and no encrypted) some thing like the link below:
https://lab401.com/blogs/academy/chameleon-mini-mifare-cracking-via-the-reader-attack
(Just please Explain in details and step by step)
3 - About sniffing : could you please explain that How can I do that?what is the results?what is the distance between the chameleonmini REV-G and the card+reader?

gtpy, 2019-01-29:
My GUI you find at gitlab  (no more available under github)
https://gitlab.com/Gtpy/ChameleonMini-RevG_GUI

geo-rg, 2019-01-29:
Hi @ictmehrdad,
regarding the attack: I cannot give a step-by-step explanation of any attack due to legal reasons. I think you can find explanations somewhere in the internet, at least you can read the publications about Mifare Classic on google scholar and use the tools from the proxmark3 github repo to recover keys.

The sniffing for both directions is a relatively new feature. The direction reader->card should be detected very well, while card->reader sometimes has problems. There is not _the_ best position to sniff, you may try out few. One idea is to place the Chameleon directly on the card such that the Chameleon is nearer to the reader.  In order to use the sniffing feature, you have to set `logmode=memory`. After finishing, you can download and view the log via the chamlog.py tool provided in the Software folder of this repository. You can also use this to view the communication when the Chameleon emulates a card.

ictmehrdad, 2019-01-29:
Okay
then Can You give me the answer of below Q:
1 - Can I have a reader attack with ChameleonMini REV-G (Not Proxmark3 Rebooted)
2 - I download the ChameleonMini REV-G from github;But I don't Know what should I do to run that Code to see the GUI; Can You help me to do that?
3- The Tools that I need to crack the Key after read the reader with ChameleonMini REV-G are like Proxmak3 Rebooted?or I need another tools?

ictmehrdad, 2019-01-29:
I want To know that what is the best GUI software that we can Use to work wit that?
Should I use the Proxmark3 Rebooted Software for ChameleonMini REV-G or not?
and In Total I want to know your Idea About this To Products?I mean ChameleonMini REV-G and Proxmark3 RDV4.
Thanks

gtpy, 2019-01-29:
I think you're getting a little confused here.  Proxmark3 rebooted does not exist.   Either you mean the GUI for the RevE rebooted.  Or use my GUI for the revG.  The PM3 RDV4 is also a great tool.  I think both are good and I like to use them in combination.
Which is the best GUI I can't say.  Try it out and form your own opinion.  I find them all good in their way.

gtpy, 2019-01-29:
@ictmehrdad you cloned the firmware repo here from emsec or the one from the GUI?
And if you cloned the GUI repo which one you have cloned?

ictmehrdad, 2019-01-29:
1- Then can I use the Chameleon REV-G to read from reader and then crack it to find a key!?and make a clone card(without accessing to the RFID card, I mean I just have an access to the reader)
2- I download it from GUI!!
Don’t you have .exe or .msi file!?
3- Do you know the majority of Hotel RFID card models!?(Mifire 1k or 4k or etc)
Thanks dear

ictmehrdad, 2019-01-29:
> You could also use the terminal interface if you have problems with the gui, I'd suggest you to try it because it depends only on the firmware currently running on the chameleon.

Before to create this issue 
I use tera term and do something with it
And also update the Frimware 
I just know more about cracking and read from reader and find the key then clone

ictmehrdad, 2019-01-29:
And also I used the android apk too(it has alot of bugs,but wonderful)
I find the earlier version of that master on the web and work with that
Then I saw the project on github
Now I know how to work with terminal and gui
I want to know that can I read the reader data with ChameleonMini REV-G or not
If yes How!?
Then for cracking, Can I use the same software that proxmarke rebooted used or not!?
If No, what should I do.

ictmehrdad, 2019-01-29:
I work with all of the commands and options 
But 2 of them were a little ambiguous:
1-Sniffing (Ex: What are the data that we sniff?! Is it all of the data card?! What is it?!
I can sniff a card and then clone it but I don’t know what is it?!
I am a little confused 
2- the one that I explained

ceres-c, 2019-01-29:
>  I just know more about cracking and read from reader and find the key then clone

I've never used gtpy's gui, but if it works as Iceman's one does you should follow these steps.
https://github.com/iceman1001/ChameleonMini-rebooted/issues/36#issuecomment-387050124
Ensure you have a compatible firmware first.

On the other side I'd say that this automatic reader attack is just an automated version of a longer process which involves emulating a card without knowing the keys and passing received keystreams to the [mfkey32 tool](https://github.com/Proxmark/proxmark3/tree/master/tools/mfkey).
Additional information on how to use it [here](http://www.proxmark.org/forum/viewtopic.php?id=5052)

 > I want to know that can I read the reader data with ChameleonMini REV-G or not

You can read it, which translates into practice to emulating a mifare card and logging to memory what the reader sends you.

>  Then for cracking, Can I use the same software that proxmarke rebooted used or not!?

For cracking you can use the same mfkey32/mfkey64 that can be used with proxmark traces, yes

>  1-Sniffing (Ex: What are the data that we sniff?! Is it all of the data card?! What is it?!

That's what datasheet and standards are made for. You've got to read how the protocol works :-)

ictmehrdad, 2019-01-30:
> iceman1001/ChameleonMini-rebooted#36 (comment)

Thanks for your answer 
1- Could you please how can we find out that a card is encrypted!?
2- I face a card that in each time it has a new UID; How can Clone that or crack it?!
3- about reader attack, Is it possible to tell me the terminal commands too!?what should I Do!?you know in chameleonmini rev-g there is no MF-Detection item! Then what should I do?!

ictmehrdad, 2019-01-30:
And also Is it possible to explain the diffrent between the replay or relay attacks of ChemelonMini REV-G?
thanks

gtpy, 2019-01-30:
> My GUI you find at gitlab (no more available under github)
> https://gitlab.com/Gtpy/ChameleonMini-RevG_GUI
@geo-rg would you please correct the link in the wiki
thanks.

ictmehrdad, 2019-01-30:
Thanks
You know the new version That I use doesn’t have the MF-Detection for the REV-G Frimware.
What should I do!?
Can I use the REV-E Frimware for my REV-G to see that options!?
Or do you have a command for the REV-G to enable MF-Detection function and them found the key!?
One more thing is for Mifire 3DES and AES that I don’t have the key, even one key. What should I do ?
Is their any way that I can earn a guest card of a hotel ( with 3DES and AES)and then make a master card!? What is you solution and idea about it?!
And I have a suggestion to add MFkey64 to the GUI too
Thanks

gtpy, 2019-01-30:
Can you specify which GUI you mean? 
There are several. 

You cannot use the RevE fimware. 
You need to modify the firmware of RevG.
Feel free to contribute (and modify the firmware or GUI) and bring something back to the community (PR). You can check the RevE rebooted firmware or the PM3 repository for hints about the mfkey attack.

geo-rg, 2019-02-26:
Hi @ictmehrdad 
I assume you have found a solution. If you have any further issue related to this problem, please re-open this issue. If you have any *new* problem, please open a new issue.

closed 2019-02-26 by geo-rg

ccb3b72, 2019-04-09:
Hi Guys, 

I tried to find my question in the forum before writing this, without success... :)

I want to do a hardnested attack on my Mifare Plus card. Since I have no default key, I sniffed a transaction with ChameleonMini RevG. It seems, it worked. I could find the price of the coffie :) and also found my balance... (Maybe it is a false positive but I do not think so) 

Actually I checked a few resource/demo but it did not yield clear info how I should proceed. Could  you give me a url which describe how can I "extract" my first key I wish so much :)

Btw I found a Plus related A key in a Mifare related pdf, but it has 14 characters. Is not it weird a bit? I just can probe it a few days later.

Thanks
Richard

eXspir3, 2019-04-10:
> Hi Guys,
>
> I tried to find my question in the forum before writing this, without success... :)
>
> I want to do a hardnested attack on my Mifare Plus card. Since I have no default key, I sniffed a [...]

I am currently trying something similar - how did you manage to activate the sniffing mode and read the data?

Best Regards

geo-rg, 2019-04-10:
Hi @eXspir3, 
I replied to your question [here](https://github.com/emsec/ChameleonMini/issues/188#issuecomment-481747159).