emsec/ChameleonMini #204: SEND / SEND_RAW command not working as expected

Vrumfondel, 2019-02-01:
Hi !

It seems SEND/SEND_RAW is not working as expected.
The Bytes are sent out, but the Chameleon does not wait for any response from the card.
In the current implementation the response would have to be received within 1 call cycle of Reader14443AAppProcess() - which is kind of impossible .....

Vrumfondel, 2019-02-01:
Yes I autocalibrated.
Commands like IDENTIFY and GETUID work as expected, for SEND I receive "NO DATA" right away.

I tried the following:

In Reader14443AAppProcess(), in the handler for Reader14443_Send and Reader14443_Send_Raw I commented out the part for Bitcount == 0, and put the rest to an "else" clause (sure this is not a clean solution).
In this case I get responses from the Card, which leads me to the conclusion that the time between two calls of Reader14443AAppProcess() is too short to receive the response from the card (this is what I meant to say with "call cycle".

geo-rg, 2019-02-26:
Hi @Vrumfondel 
`send` and `send_raw` are special commands. On the one hand, they are timeout commands, because their function cannot be handled by one call of the `CommandExecParamSend` function. On the other hand, if after `Reader_FWT` ms there is no answer, the command knows that there is no answer. So yes, setting the timeout does not change anything for this function.

I really don't know what's the problem for you, since I cannot reproduce the probleme here. I flashed both the "Latest" firmware and the compiled the current master branch and both worked out of the box.

Vrumfondel, 2019-02-28:
OK, then maybe let's try differently: Can you give me an example of an raw command for a Mifare Classic card which works for you - then I can doublecheck on my side.

geo-rg, 2019-02-28:
With *any* ISO14443A card, `send 26` should give a response every second time. If you issue `send 9320` after receiving an answer from `send 26`, this should give you `9370` followed by four bytes of the UID and a check byte (and maybe 2 bytes CRC). `send_raw` should work analogously.

Vrumfondel, 2019-02-28:
OK, this works. Then the problem must be the commands I send.
Thanks for the support 1

closed 2019-02-28 by Vrumfondel