emsec/ChameleonMini #207: New Chameleon setup

ph553, 2019-02-22:
Hi
I just received my chameleon rev G and i flashed the firmware as instructed. I am using iceman's GUI and this is the main screen i get:
[image]

ph553, 2019-02-22:
Nevermind, i tried using the serial interface and it's working.

closed 2019-02-22 by ph553

gtpy, 2019-02-22:
Maybe try https://gitlab.com/Gtpy/ChameleonMini-RevG_GUI

-------- Original-Nachricht --------

ph553, 2019-02-22:
Thanks, i'll try it :)

gtpy, 2019-02-22:
But I wonder why the iceman gui not work. I test it a few weeks ago and it worked. I'll try it out again later today.

-------- Original-Nachricht --------

ph553, 2019-02-22:
Does the iceman program have a cache? i used to have a rev E rebooted and it might be because the program thinks it's still a rev E?